winder/Universal-G-Code-Sender #1757: Outline goes too fast

chrisla23, 2021-12-27:
**Is your feature request related to a problem? Please describe.**

When I use the "outline" functionality to confirm my cut will be where I anticipate on the material, it goes too fast.  I would also like to see it lift up on the Z-axis before starting, to avoid any potential crashes. 

In addition, a nice enhancement would be to have it pause or decelerate briefly when it reaches the most distant extent of each axis. 

**Describe the solution you'd like**

I would like it to either go slower or be configurable. A fixed-configuration setting would be preferred, failing that having it generate the gcode at a speed matching the current jog speed. 

Thanks,

-Chris

breiler, 2022-01-13:
Thanks for the suggestion, I have changed so that it uses the jog feed rate instead of the max machine speed.
I am not sure about messing with the Z height though, I need to think some more about that.

chrisla23, 2022-01-13:
Nice! I may have another potential outline bug for you. I still need to make sure I can accurately reproduce it and take some screenshots.  

Any thoughts on having it pause briefly at the extents? 

Thanks,

-Chris

breiler, 2022-01-13:
Regarding the pause or deceleration on the extents is a bit more tricky. There are a couple of functions in the toolbox ([JTS](https://locationtech.github.io/jts/javadoc/index.html)) that could be used for that but would require a lot more work.

marshr32, 2022-01-17:
@chrisla23 
As a work around for pausing at the corners you could right click each corner and "jog to here". 
It's more work of course, but would get the result.

chrisla23, 2022-01-17:
In cases where there is an excess of material yes, but when it was very close to the material edge combined with the visualizer needing improvement that becomes risky.  If I knew where "here" was within a 1/8th or so that would work, but I really can't with the state of the visualizer today. 

Having the speed tied to jog speed will help immensely, this would have just been an extra. 

-Chris