winder/Universal-G-Code-Sender #1758: Cant start miling g-code imported from flatCAM i think it is M6 error

oliskcz, 2021-12-27:
Heres the g-code. Thanks in advance
[g-code.txt](https://github.com/winder/Universal-G-Code-Sender/files/7781434/g-code.txt)

breiler, 2021-12-28:
UGS does not support the M6-command (manual tool change) so it needs to be removed from the gcode.

closed 2021-12-28 by breiler

oliskcz, 2021-12-28:
How to remove it i dont want to mess up the gcode.

khturnings, 2021-12-28:
I found this useful regarding this issue - although it does not say how to remove M6 from gcode.

I think it depends on how you generate your gcode - I use Fusion 360 to generate my gcode and you can get it to suppress Tool change (not a good idea going by this article https://www.3dtek.xyz/2020/01/29/grbl-m6-errors/ )
[image]

Else use NotePad++ or something similar and use find and replace - and then do test air cut.