winder/Universal-G-Code-Sender #1759: UGS java error

oliskcz, 2021-12-28:
### **Hello there!**
I have another issue.
**ISSUE:** So when i open gcode the editing window for it doesn't open and it gives me one error here is is:
[image]
And heres the gcode: [g-code.txt](https://github.com/winder/Universal-G-Code-Sender/files/7785177/g-code.txt)
Thanks.

breiler, 2021-12-28:
Opens fine for me, what version of UGS are you running?
Can you post your log file: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

oliskcz, 2021-12-28:
Im using ugs 2.0.9.
I cant find the logfile.
When i start milling the motor is not turning when i deleted the M6 (my motor is not burned or anything.

And this: [image]
[image]
[image]

oliskcz, 2021-12-30:
Can you help me?

breiler, 2021-12-30:
The screen shots you posted about warnings in the editor is because the editor highlighter doesn't recognize commands with padded zeros (G0 -> G00). I have made a fix for it but it will take some time for me to getting it merged. But those are just warnings, so you can ignore those.

It is really difficult to help without the log file. You could press the button "Show Details" and paste that content here which might give a clue.

chappy1978, 2022-01-11:
Im getting the same issue when I connect my GRBL for the first time. 
Im using UGS 2.0.9, MacOS, GRBL-MEGA

java.lang.NullPointerException
	at com.willwinder.universalgcodesender.model.GUIBackend.canSend(GUIBackend.java:662)
	at com.willwinder.ugs.nbp.core.actions.StartAction.isEnabled(StartAction.java:75)
	at com.willwinder.ugs.nbp.core.actions.StartAction.lambda$UGSEvent$0(StartAction.java:70)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
[catch] at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:136)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/7845891/messages.log)

chappy1978, 2022-01-11:
I when and used the nightly build and that has cleared the Exception.

breiler, 2022-01-11:
@chappy1978 thanks for the log. The errors in your log file is a known issue and a fix is on its way. But that should not cause the errors described by @oliskcz:
> gcode the editing window for it doesn't open

> When i start milling the motor is not turning when i deleted the M6 (my motor is not burned or anything and it works on other g-codes other than from flatCAM.)

Do you have the same problems as @oliskcz with it not opening or sending the file?

breiler, 2022-01-14:
@oliskcz, @chappy1978 I have made some updates that should resolve this issue. Could you try the nightly build and see if things are working better for you?

breiler, 2022-03-22:
Closing as we did not get a response from users.

closed 2022-03-22 by breiler