winder/Universal-G-Code-Sender #1763: problem ugs with java

priandr, 2022-01-07:
**Describe the bug**
hello i have a problem with ugs. I download version for windows. I have windows 7 64 bit. When i open and connect the machine cnc if i move by jog controller axis x, y, or z in many cases i have an error unexpecting with java java.lang.NullPointerException. I have the same problem when I open file gcode with a work. In many cases it start works bat not finished. For the same error. If you can halp me
Thanks.
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/7828685/messages.log)

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
[e.g. UGS Platform 2.0.6, UGS Classic 1.0.9]

**Hardware**
[e.g. GRBL 1.1f, TinyG,  SmoothieBoard]

**Operating system (please complete the following information):**
 [e.g. Windows 10, MacOSX, Ubuntu 20.04 ]

**Additional context**
Add any other context about the problem here.

breiler, 2022-01-07:
Yes, this is an issue that sneaked in in the latest release and has been partly fixed in the nightly release.
Once connected this error should not cause any problems and the gcode programs should run fine. 
What controller are you using? (ie. Grbl 1.1)

priandr, 2022-01-07:
Grbl 1.1 h. But i can't work. If i move axes with jog controller movement were blocked. If I open a gcode it begins but after a few seconds it stops. We can resolve?

breiler, 2022-01-08:
Can you post a log where you have tried jogging and when you start a gcode file?

SteveSamScott, 2022-01-08:
Hi, I think i may be having the same issue (sorry I'm not too savvy on computers) and dont know how to "post a log"... I was able to get a screenshot  if thats any help.
[image]

'

SteveSamScott, 2022-01-08:
I have windows 
[image]
7 64 bit. When i open and connect the machine cnc if i move by jog controller axis x, y, or z in many cases i have an error

breiler, 2022-01-08:
@SteveSamScott you can find the log files here described here https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files. When the problem occur, clues on what you tried to do and any error that may have happened should be visible there and would be helpful to figure this out.

SteveSamScott, 2022-01-09:
@breiler , Thank you. Ive found 2 which may help (i had to save them as a txt file in order to upload them, i hope thats ok.
At the time, id newly installed the prg. and was trying to jog. when the machine failed to respond, I tried to configure a new machine and got confused by the scant instructions at 'Calibration'. The machines individual axis began moving at extremely slow pace, creeping along as the dialogue box increased in .1mm intervals. Not something I remember from experienced with a previous version on my old (now dead) computer.
 I believe my calibrations are fine as they are currently accurate when running VTransfer (a Vectric G-code sender for Aspire).
Thank you for your interest and support, i'd love to financially support you too once I can get back into the workforce (2yrs of C-19 has devistated my income).

[messages log1.txt](https://github.com/winder/Universal-G-Code-Sender/files/7834557/messages.log1.txt)
[Messages log2.txt](https://github.com/winder/Universal-G-Code-Sender/files/7834558/Messages.log2.txt)

breiler, 2022-01-09:
@SteveSamScott thanks for the logs, I found a problem with the Diagnostics window, close it and restart UGS.
I am not sure that this will resolve the issue, but it looks like it spewing a lot of errors so its a start.
[image]

About the slow movement in calibration, it uses the feed rate from the jog panel. Maybe a bit unintuitive and could be improved. 
[image]

Don't worry about any financially support, thanks for taking your time to help and resolve this.

SteveSamScott, 2022-01-09:
@breiler YOU CHAMPION!!
Problems solved completely.
thanks abound for your help.

priandr, 2022-01-09:
@breiler hello i've just try to do some operation but nothing. I've the same problem. I attached two log, one for the jog controllere and one at the open and play work gcode. I attached also my firmware in case something was wrong. An other question Step size ABC is 1 in my configuration. It is correcly set up?
Thanks
[messages jog controller.log](https://github.com/winder/Universal-G-Code-Sender/files/7834925/messages.jog.controller.log)
[messages work gcode.log](https://github.com/winder/Universal-G-Code-Sender/files/7834926/messages.work.gcode.log)
[doc1.docx](https://github.com/winder/Universal-G-Code-Sender/files/7834935/doc1.docx)

breiler, 2022-01-09:
@priandr could you try pressing the "Get state" button when this happens:
[image]

priandr, 2022-01-09:
Thanks for all

[doc1_2.docx](https://github.com/winder/Universal-G-Code-Sender/files/7835762/doc1_2.docx)

breiler, 2022-01-09:
Thanks for the detailed screenshots!

This makes me suspect that your connection to the GRBL controller is unstable:
[image]

I would check that the USB cable is properly shielded and that it is placed far from any power lines such as to the spindle.
Does it work if you run the gcode program without powering the stepper motors or spindle?

You could of course try with another gcode sender, here is a list to start with:
https://github.com/winder/Universal-G-Code-Sender/wiki/Software-comparison#gcode-sender