winder/Universal-G-Code-Sender #1779: Please separate Z jogging distance settings from X and Y settings

cdo1955, 2022-01-18:
### Description

Please separate the Z jogging distance setting from the X and Y setting. So Z does not change when  X and Y are set. Problem is when clicking the "Larger" setting for X and Y , the Z setting can jump up to 100 mm. Which can drive the Z axis into a crash. The way it is now, I'm constantly changing the Z setting back to 1-10 mm after making a change to X and Y axis. That's an unsafe way to have it set up. The Z axis jog settings should be totally separate from the X,Y settings.

chrisla23, 2022-01-19:
It already does this. Tools->Options->UGS->Sender Options->Use separate step sizes for Z and XY Jog Movements

-Chris

rspeed, 2022-08-11:
This can be closed.

breiler, 2022-08-11:
I think that what @cdo1955 means is to have separate buttons for increasing the step size on different axes:
[image]

I still think that this is a valid feature request.

cdo1955, 2024-02-16:
Separate feed rate for Z jogging? The idea is similar to my first request, after jogging at high speeds and feeds on X and Y, you can easily jog Z with the same high feed rate that was used on X and Y. Which could lead to a crash. It would be best if Z had a separate feed rate along with its separate speed.