winder/Universal-G-Code-Sender #1781: Figure out how settings should be stored between different versions

breiler, 2022-01-19:
### Version

Other

### Hardware / Firmware

Other (add details about it below)

### What happened

When upgrading from one UGS version to another, how should the settings be transferred?
See discussion here: https://github.com/winder/Universal-G-Code-Sender/discussions/1755

marshr32, 2022-01-29:
The latest nightly seems to do a great job at picking up my settings.
I guess that could be because I set up a nightly previously and it's looking at the same config file?

breiler, 2022-01-29:
Yes, it uses the same settings directory between all nightly builds. We could do the same for the stable builds but this requires some testing.

breiler, 2022-12-15:
Now uses the same settings directory. Fixed with #2038

closed 2022-12-15 by breiler