winder/Universal-G-Code-Sender #1782: No option to continue after "door" event

marshr32, 2022-01-20:
### Version

2.0.10

### Hardware / Firmware

GRBL 1.1

### What happened

I use the door input as my pause button because it stops the spindle.
In 2.0.9 I can continue afterwards, but not 2.0.10

### How to reproduce

2.0.10 log
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/7902445/messages.log)

### Operating System

Win 10

breiler, 2022-01-20:
Thanks for the report, the issue should be resolved in the latest nightly build and I would appreciate help testing this.

marshr32, 2022-01-24:
Hi sorry for the delay.
Have tested with nightly 22-01-20 and this issue is resolved. 
However its revealed another, more major bug so I'll raise a report for that.

closed 2022-01-30 by breiler