winder/Universal-G-Code-Sender #1786: Machine continues after "door" then "stop"

marshr32, 2022-01-24:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

Started G-code program and realised Z zero was wrong, bit had hit workpiece when it should have been at clearance height. Not too sure how this happened as there was clearance before I hit send, it's like the Z moved the wrong direction...... but anyway.....

So I hit the stop button on my controller (PS3) which triggers the "door" function to pause the program. I then hit the STOP button on the UI on the laptop to reset everything so I could jog Z out, and the program kept going! 
This caused it to do a rapid move with the bit still trying to move into the workpiece..... result a snapped bit and gouge in the workpiece (see pic). Spindle wasn't going as it was supposed to be cutting air.

I think it did whatever move was still in memory/buffer.

It's reproduceable by either doing the above or by moving any jog stick on the controller while in "door" mode.
Jog controller movements should have no effect on the running program IMHO. 

As a side note, jog controller movements will also cancel a running program, not sure this is a good idea either.

If I hit the E-stop after "door" it cancels everything and resets. I think the stop button on the UI should do the same.

It doesn't continue when pause is hit on the UI then UI stop, but I don't use pause as it won't turn off the spindle so I'd have to do that manually.

[image]

### Operating System

Win 10

### Anything else

Message log attached, I did the same a few times to test, all same as above.
[messages1.log](https://github.com/winder/Universal-G-Code-Sender/files/7922478/messages1.log)
.

breiler, 2022-01-24:
That is painful to see, I am so sorry. 
I thought that I did a lot of testing on this, but clearly not enough. 😢

breiler, 2022-01-26:
I have a fix for this in the latest nightly.
When hitting the "Open door"-action it will pause the machine and enter the door state. Pressing play resumes the machine after spinning up the spindle. Pressing stop will clear the buffers and reset the controller (that was the only way to exit the door state to stop).

marshr32, 2022-01-29:
Apologies for the delay, I've been away.

This seems to work great now, and the reset is absolutely fine.
Thanks for implementing it so quickly.

Quick question, is it possible for UGS to make the Z retract during door events?
I know its possible in GRBL generally, but not with my machine as it's coreXY, so this config doesn't fit in memory.

breiler, 2022-01-29:
I am the one that would need to apologise, I still feel bad for almost destroying your machine...

No I don't think that retracting would work. When entering the door state, the controller may have a bunch of commands in the planner buffer that is waiting to be executed. When we start the feed again it can resume executing the commands in that buffer. For us to be able to send the retract command we would need to clear the buffer and then we wouldn't know which commands to resend. 

So this would need to be implemented by the controller I believe.

marshr32, 2022-01-29:
All good, $hit happens and it was only a cheap bit anyway. The machine is large so it'll be fine.

I thought that would probably be the case with retraction, but worth an ask. 
I'll have to look further into implementing in GRBL.

closed 2022-01-30 by breiler