winder/Universal-G-Code-Sender #1800: [Error] Error while processing response <<Idle,MPos:0.000,0.000,0.000>>

bullstreetboy, 2022-02-07:
### Version

2.0.9

### Hardware / Firmware

GRBL 1.1

### What happened

After opening the port and getting the $$ listing in the console I get infinate:-

[Error] Error while processing response <<Idle,MPos:0.000,0.000,0.000>>

Show verbose output is not checked.

GRBL Controller is driving a cnc 3018 clone and works with Candle/Lasergrbl etc.

### How to reproduce

Rebboted everything and same result after opening port.

### Operating System

Windows 10

### Anything else

This has only just started to happen after using 2.0.7 for weeks. So installed 2.0.9. same thing.

I did reflash GRBL 1.1 using Arduino 1.8.16 with board = "Arduino Duemilanove" and processor "ATMega328P" - uploaded OK.

breiler, 2022-02-07:
Could you post the log file: `C:/Users/[your username>]/.ugsplatform/2.X/var/log/messages.log``

bullstreetboy, 2022-02-07:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/8015800/messages.log)

bullstreetboy, 2022-02-07:
I'm sorry,  but I don't understand "Fixed problem with NPE ....."

breiler, 2022-02-07:
Sorry about that, that was just github messaging about changes related to this bug report.

I found an issue in the log that you provided and created a fix for it which is done compiling. Can you try the latest nightly build from here: https://github.com/winder/Universal-G-Code-Sender#downloads

bullstreetboy, 2022-02-07:
That fixed it - Thank you very much
Amaxing response and fix time  - very impressive!

closed 2022-02-07 by breiler