winder/Universal-G-Code-Sender #1802: UGS built-in pendant will not operate via wifi connected phone if computer is on LAN

i-g-g-y, 2022-02-08:
### Version

2.0.9

### Hardware / Firmware

GRBL 1.1

### What happened

I am unable to access the built-in pendant via phone. The PC is on LAN, in this case it works flawlessin its own browser, but not on the wifi connected phone. I decided to pull the LAN out and connect PC to wifi and the pendant feature still does not work on the phone. If accessing the given url on PC it still works, but with glitches (no connection initially, but then after some time it starts).

### Operating System

WIN10

### Anything else

Occurs every time.

i-g-g-y, 2022-02-08:
Also it issues 3 distinct IPs to connect to:
Pendant URL: http://169.254.192.27:8080Pendant URL: http://192.168.12.200:8080Pendant URL: http://192.168.56.1:8080
[image]
When checking URL on desktop it says not connected:
[image]
[image]

breiler, 2022-02-09:
First off, you are running an ancient version of UGS and you should probably upgrade.  You are not running version 2.0.9 which you specified in the issue.

The three IP-addresses means that you have three different LAN-adapters installed on your computer. And how this should work depends entirely on how you have set up your network. 

http://169.254.192.27:8080
This network adapter is probably not connected, or connected to a network without a DHCP-server and should probably be ignored

* http://192.168.12.200:8080 
* http://192.168.56.1:8080
These IP-addresses are in different series which indicates that you have two different routers running? You need to make sure that the clients on the different networks may access each other.

Do you have a firewall on the PC that restricts incoming traffic?