winder/Universal-G-Code-Sender #1807: Raspberry Pi based on Debian Bullseye

QTronics, 2022-02-16:
### Version

2.0.9

### Hardware / Firmware

GRBL 1.1

### What happened

Trying to install on a new raspberry pi with the lasted version of Raspberry PI OS (64 Bit). Build Date 2022-01-28. UGS Platform will not open after installation. 

### Operating System

Raspberry PI OS (64 BIT) Debian Bullsye 2022-01-28

### Anything else

I also tried the 32 bit version with the same issues.

breiler, 2022-02-16:
Can you post log files located at `~/.ugsplatform/2.X/var/log/messages.log`

QTronics, 2022-02-16:
That path does not exist on the pi. 

The path is /home/pi/ugs/ugsplatform-pi

I searched for a messages log file on the system and could no find one.

QTronics, 2022-02-16:
I Used the instruction from here.
https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi
I could not find gl2 fake in raspi-cofig. 
Apparently it is not present on the latest version and is supposedly enable by default now.

breiler, 2022-02-16:
What do you mean by `I searched for a messages log file on the system and could no find one.`. 
Searched how and where?

The configuration and logs are placed under a hidden directory under `~/.ugsplatform`. 
Does that folder exist?

If you start the software from the terminal, does it print anything?

QTronics, 2022-02-16:
pi@raspberrypi:~/ugs/ugsplatform-pi/jdk/jdk-13.0.1+9/bin $ ls -l
total 460
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jar
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jarsigner
-rwxr-xr-x 1 pi pi  8632 Oct 29  2019 java
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 javac
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 javadoc
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 javap
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jcmd
-rwxr-xr-x 1 pi pi  8660 Oct 29  2019 jconsole
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jdb
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jdeprscan
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jdeps
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jfr
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jhsdb
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jimage
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 jinfo
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 jjs
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 jlink
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 jmap
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jmod
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jps
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 jrunscript
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jshell
-rwxr-xr-x 1 pi pi  8648 Oct 29  2019 jstack
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jstat
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 jstatd
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 keytool
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 pack200
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 rmic
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 rmid
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 rmiregistry
-rwxr-xr-x 1 pi pi  8640 Oct 29  2019 serialver
-rwxr-xr-x 1 pi pi 89728 Oct 29  2019 unpack200
pi@raspberrypi:~/ugs/ugsplatform-pi/jdk/jdk-13.0.1+9/bin $

QTronics, 2022-02-16:
pi@raspberrypi:~/ugs/ugsplatform-pi/jdk/jdk-13.0.1+9/bin $ cd ~/.ugsplatform/
bash: cd: /home/pi/.ugsplatform/: No such file or directory

breiler, 2022-02-16:
Which version and variant did you download?

I know that we switched back to Java 8 because of some performance issues on ARM-based processors. So you are either running an old version (because I see Java 13 in your previous posts), or the wrong variant, or I screwed up the packaging...

Make sure you download the one marked for RaspberryPI as it's using Java specifically for ARM:
https://ugs.jfrog.io/ugs/UGS/nightly/ugs-platform-app-pi.tar.gz

QTronics, 2022-02-16:
I Just tried that and get a similar issue. 
pi@raspberrypi:~/ugs/ugsplatform-pi/bin $ bash ugsplatform
./../platform/lib/nbexec: line 430: /home/pi/ugs/ugsplatform-pi/jdk/jdk8u275-b01-jre/bin/java: No such file or directory
pi@raspberrypi:~/ugs/ugsplatform-pi/bin $

QTronics, 2022-02-17:
I ended up going back to the buster distro for raspberry pi and it works fine on that.  I could not get it to work on bullseye.  Thanks for your help.

closed 2022-02-17 by QTronics