winder/Universal-G-Code-Sender #1816: Ability to test probe circuit before initiating probe macro

bryonmadge, 2022-03-10:
### Description

I would like to see an "Led" indicator on screen to test the continuity of my test probe each time before committing to initiating the probe routine. 
I use Mach3 on another machine and I've gotten used touching my probe wires together to quickly test for continuity before running. Mach3 has and on-screen indicator for this.
Thanks

chrisla23, 2022-03-10:
All alarms including the probe will show up in red when activated. This happens at the bottom of the DRO pane. Depending on your window size you may need to scroll down some to see them. 

I do this same test you are suggesting each time I use the probe. 

Thanks,

-Chris

bryonmadge, 2022-03-11:
It was hiding and I see it now. Thanks.

closed 2022-03-11 by bryonmadge