winder/Universal-G-Code-Sender #1826: Describe the feature

daxliniere, 2022-03-19:
### Description

If you have a macro assigned to a Toolbox button, renaming that macro causes glitches in the Toolbox. The macro needs to be re-assigned to a Toolbox button.

breiler, 2022-03-21:
Thanks for the feedback.

It was originally like this but had to be replaced with a popup because the text input fields could not be styled in a good way that was consistent across different OS:es.

There might be another solution where we could temporarily replace the JLabel with a JTextField when it is clicked. But this change will go far down on my priority list though.

daxliniere, 2022-03-21:
Hey Joacim! Thanks for taking the time to reply.

Ahh, I see what you mean, but I think we're talking about _slightly_ different things. Though it might be more 'slick' to edit the DRO display directly, I don't have any problem with the popup text box.
And to that end, I was thinking you might be able to add a function that simply selects all of the editable text in that box upon popup. Essentially, just add a CTRL+A macro as soon as that dialogbox, then we can type right over it.

breiler, 2022-03-21:
🤦 Sorry

Great feedback, should be a quick fix!

daxliniere, 2022-03-21:
haha, all good, man! Thank you for developing this software and let me know if I can help in some way. :)

breiler, 2022-04-09:
The build server is currently having problems with some tests, but once its up and running again this should be available in the nightly build.

closed 2022-04-09 by breiler

daxliniere, 2022-04-20:
Oh man, this is really fantastic. I know it's a tiny little thing, but it's a big workflow improvement!