winder/Universal-G-Code-Sender #1828: Allow endpoint switches to reside outside work area.

i-g-g-y, 2022-03-19:
### Description

Hello.

I would like the end switches to reside outside the work area. Right now the work area includes <0;0;0>, meaning that if we want to use whole area, the switch will trigger. I know this is a bit lame, as we never use the entire work area, but it gets annoying when jogging and hitting the limit switch in work space.

breiler, 2022-03-21:
This sounds like something that should be configured in the controller or where your physical limit switches are placed. 
UGS is dumb in this regard, just reporting the current controller position and relying on the firmware to throw errors if any movement is considered valid or dangerous.

i-g-g-y, 2022-03-21:
I undrstand. Thank you!