winder/Universal-G-Code-Sender #1831: Function to draw Visualizer based on program line number

daxliniere, 2022-03-20:
### Description

I just woke up this morning an idea. It would be amazing if the Visualizer could display the block you are up to based on which line is selected in the program source window (the Editor).

Especially for people like me who are starting out, it would be invaluable to be able to see exactly how each command relates to a movement.

A button in the toolbar would be an easy way to toggle this function.

All the best,
Dax.

breiler, 2022-03-21:
This feature does exist (if I understood your request) and has been improved quite recently and is a part of the latest nightly build. There is a screen shot of these changes here: https://github.com/winder/Universal-G-Code-Sender/pull/1764

daxliniere, 2022-03-21:
Yep, you've got it!

Would be nice to have an option (with toolbar button) to enable the behaviour I suggested above, though.
I think that when I used the word 'selected', it was in error. A better description would be whatever line the text cursor is on.

closed 2022-04-20 by daxliniere