winder/Universal-G-Code-Sender #1844: Console & Visualizer Freezing ~30 commands into job

caswal, 2022-03-26:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

After starting a job, after about 20-40 commands or so. The Sent status becomes stuck "Send Status: (18/357)". The console stops updating, and the visualiser no longer updates either. The time/estimated time still updates.

Commands are still being sent in the background, so the job completes, so nothing is lost. Which is nice.

Stop, Pause buttons no longer work. Can Disconnect, but can't reconnect. 
Program can be "Closed" but remains open in background. Still holds lock on the comm resource and must be killed in task manager.

Log Attached. Occurs everytime on this setup.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/8356496/messages.log)
[SysInfo.txt](https://github.com/winder/Universal-G-Code-Sender/files/8356504/SysInfo.txt)

### How to reproduce

Occurs every time on this setup. Just start a job, and occurs within the first 40 commands or so.

### Operating System

Attached msinfo32 dump above. Win10 Build 19042. HP Thin Client T630 4gb ram.

chrisla23, 2022-03-26:
Have you tried increasing your java heap size? 

Linking to this as it is a well-written article, not CNC specific:

https://www.bobscnc.com/blogs/troubleshooting/troubleshooting-increasing-ugs-memory

-Chris

breiler, 2022-03-27:
Thanks for the the great bug report. Like @chrisla23 mention it seems that the software has run out of memory. 
I can see that you are rotating the model (left then right), does this still happen if you don't do that?

The article that @chrisla23 linked to shows you how to set the minimum size (the amount of memory that UGS should start with). You could try to increase the maximum size from 512Mb to 1024Mb by adding `-J-Xmx1024m`

caswal, 2022-04-02:
Thanks for the prompt reply. Weekend again so have a play.

Using the bobscnc guide, increased the memory from the default 64m in the conf up to 1024m. Task manager reports it using 750mb. I am using the 32bit version. I'll give the 64bit a go tomorrow.

caswal, 2022-04-02:
A bit more machining this morning. I redact my previous statement. No improvement, is still generally freezing 20-40 commands in.

breiler, 2022-04-09:
Did you set the `-J-Xmx1024m`? 
The instructions in the bobcnc guide are wrong...

hovercraft-github, 2022-05-02:
@caswal Can you report the JVM build used please? - it can be seen via UGS About window.
For me, similar issues occur when I run UGS with any JVM build but AdoptOpenJDK-13.0.1+9

breiler, 2022-05-02:
@hovercraft-github it was run using the bundled Java:
>Log Session: Sunday, March 27, 2022 at 11:54:35 AM New Zealand Daylight Time
>System Info: 
  Product Version         = Universal Gcode Platform 12.2-d73d3dd898acc440c460924dabf8ad5feb90369f
  Operating System        = Windows 10 version 10.0 running on x86
  Java; VM; Vendor        = 13.0.1; OpenJDK Server VM 13.0.1+9; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 13.0.1+9
  Java Home               = C:\ugsplatform-win\jdk\jdk-13.0.1+9-jre

hovercraft-github, 2022-05-02:
@caswal Yes, this JVM should be alright.
Just for curious: can you reproduce this issue when Visualizer window is closed?

caswal, 2022-05-02:
Apologies, I 'fixed' it by replacing windows with Ubuntu Mate.

breiler, 2022-05-02:
Thanks for the feedback.
Closing as we couldn't reproduce the error.

closed 2022-05-02 by breiler