winder/Universal-G-Code-Sender #1847: Options->Filter results don't include Available Actions in UGS->Toolbox

daxliniere, 2022-03-29:
### Description

Title says it all. :)

breiler, 2022-03-30:
One scenario that does not work is that it should show the sub tabs such as the following.
This should be possible to implement given the existing API:s in the Netbeans platform we are using.
[image]

---

But if I interpret the title you want to be able to filter for the action that is assigned like this?
[image]

This is as far as I know not possible with the API:s in Netbeans. =(

daxliniere, 2022-03-31:
Ahh okay, no worries, Joacim. Thanks for the reply. :)

closed 2022-03-31 by daxliniere