winder/Universal-G-Code-Sender #1848: "Program has changed, do you want to reload?" only works once

daxliniere, 2022-03-29:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

If you load an NC file, then change and save it, UGS will ask if you want to reload it. If you update the file again, UGS won't ask again if you want to reload. In fact, tonight UGS even crashed when I did this.

It would be great if this was fixed, but even better; could we please get a reload button, please?

### Operating System

Win10x64 Pro N

daxliniere, 2022-03-29:
Lastly, a related question, it seems that editing an NC program in the Source window doesn't actually change the program that UGS executes. Is that correct??

daxliniere, 2022-04-02:
> Lastly, a related question, it seems that editing an NC program in the Source window doesn't actually change the program that UGS executes. Is that correct??

@breiler, I worked out what's going on here. I tried pressing CTRL+S and it then worked as expected. The trouble is that there is no save button in the editor and more importantly, no indication that a file has been changed, therefore I assumed it was an immediate change.

breiler, 2022-04-13:
I have done some testing. The question to reload the file only happens if there are external changes made and there are unsaved changes in the UGS editor. If you choose yes then there is no local changes anymore. If the file is changed externally after this the file just reloads in UGS.

So I think that its i working as intended.

daxliniere, 2022-04-17:
Hey Joacim,

> If the file is changed externally after this the file just reloads in UGS.
> So I think that it is working as intended.
Perhaps, but I think the problem is that, once a design gets to this stage, we are often making very subtle changes that are not necessarily visible in UGS's visualiser.

And if I make a change in, say, Fusion 360 and repost the NC file, not getting a popup every time is a bit unnerving, as I'm not sure if UGS didn't detect the change or it detected the change and didn't tell me.

My vote would be for the 'file change detected' to be displayed every time. It seems like the fail-safe option to me.

chrisla23, 2022-04-17:
I recently started adding a header to my gcode in the post-processor.  In my case, vCarve/Vectric.

Eg. 

; Date: Saturday April 02 2022 10:26 PM
; Product: VCarve Pro 11.009
; File Name:  test8
; Toolpath:   Cutout Profile 1
; File notes: 
; Toolpath notes: 
; Work Piece Size: X 15.000, Y 24.000, Z 0.750
; Safe Z: 0.200
; Z Reference: Material Surface
; XY Reference: Bottom Right Corner
; Origin Offset: X 0.000, Y 0.000
; Home Position: X X0.0000, Y Y0.0000, Z Z0.2000
; Tool name: #46177-K SC COMPRESSION SPIRAL 1/4" DIA.x1/4" SHANK
; Tool notes: First pass must be at least .56 to get below the upcut portion.
; Tool diameter: 0.250

It looks like Fusion can do something similar per page 99 of this doc:

https://cam.autodesk.com/posts/posts/guides/Post%20Processor%20Training%20Guide.pdf

It also helps catch other mistakes like using the wrong tool, mixing up two similar jobs etc. 

-Chris

daxliniere, 2022-04-18:
That's a good idea, @chrisla23. I don't suppose you'd share the code for adding Date, Z Reference and XY Reference, would you? (besides going through 267 pages of that document. 😆 )

@breiler, I still think it would be a good idea to implement the change I suggested above Chris's post. ;)

breiler, 2022-04-18:
I made some more digging and found that this is actually not working as intended. We listen for external file changes and always reload the file in the visualizer and will also be the file that gets sent to the controller. 
If the user answers "No" to the question "reload file", the file you see in the editor will not be the one that gets sent to the controller... So this is pretty bad. 

Regarding changing the behaviour to always pop up the question to reload the file in the editor on external changes is built in to the Netbeans API:s and can be found here: https://github.com/apache/netbeans/blob/a1abbda4f653c801d475b5fed987ba5df7e3a35e/platform/openide.text/src/org/openide/text/CloneableEditorSupport.java#L2232
I don't see any easy way to override this behaviour, but we need to investigate this further how to properly integrate the editor.

daxliniere, 2022-04-18:
Thanks Joacim.

> Regarding changing the behaviour to always pop up the question to reload the file in the editor on external changes is built in to the Netbeans API:s and can be found here: [...]

Hmm, but surely there must be some message that is sent by Netbeans each time the file is externally changed, right? Is it not possible to use that trigger to display the warning message again?

breiler, 2022-04-18:
This functionality is buried deep into Netbeans API:s...
We might be able to intercept the event of a file being updated externally, but I'm not sure if we can intercept it before  Netbeans has updated editor and stop it from reloading automatically.

daxliniere, 2022-04-18:
Oh, no, I **_definitely_** want it to update every time. I'd just also like to know that it has happened. :)

daxliniere, 2022-04-18:
Just so I can double-check, are you saying that it will **_always_** be updated when the external file is changed?

daxliniere, 2022-06-05:
This is still an issue. I haven't tested that latest nightly build, but also couldn't see anything in the changelog about this.
To re-iterate, UGS doesn't always detect if the file has been updated. It seems like it shows a dialog box the first time, updates silently the second time, but does not update after that.