winder/Universal-G-Code-Sender #1849: intermittent unwanted jogging after release of jog button

bertieconfundo, 2022-03-30:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

Sometimes after holding down the jog button for some amount of time on a long jog, releasing the jog button will cause the machine to stop, and then continue jogging for a inch or two more, potentially leading to a machine crash.

### How to reproduce

home the machine, and then jog diagonally using the GUI buttons until the machine is about in the middle of the work area, and release the jog button. Jog stops, and then starts back up again for a bit more.

### Operating System

MacOS Monterey 12.3

### Anything else

This happens almost all the time on long jogs, especially diagonal jogs.  The reason is the algorithm that is employed in ContinuousJogWorker.java. When the jog is released and isRunning is cleared, the call to waitForCommandsToComplete() does nothing because isRunning is no longer true. Additionally, the step size isn't being calculated correctly for jogs that move in more than one axis, causing the command buffer to slowly fill up until it's completely full, which makes it more likely that when a command is sent, it's not immediately acknowledged. When the jog is canceled, the cancel jog is sent stopping the machine, but then additional buffered commands seem to be executing, causing the unwanted movement.

Additionally, when you release a continuous jog (long press), JogTopComponent.java is sending a single-step jog as well as it sees the release of the long jog as a click event.

I've worked on rewriting the algorithm in ContinuousJogWorker.java, as well as fixing the spurious extra jog in JogTopComponent.java, and I'll submit a pull request with the fix.

breiler, 2022-04-19:
@bertieconfundo, should we close this?

dvsingletary, 2022-04-19:
Yes! this is fixed.

closed 2022-04-19 by breiler