winder/Universal-G-Code-Sender #1852: Editor - modified and saving NC programs

daxliniere, 2022-04-02:
### Description

The editor's UI has a bit of ambiguity that could be easily solved.

When a program is modified, there is no indication. Traditionally the filename would have an asterisk appended to indicate this state (filename -> filename*).

There is also no save button in the default editor toolbar, which further compounded the misunderstanding I experienced, explained in the comments of this post: https://github.com/winder/Universal-G-Code-Sender/issues/1848

breiler, 2022-04-13:
The file name will be bolder when there are changes in it. However this might be hard to see as we only allow one file open at the time and there might not be any other title to compare with.

I added a save-button to the toolbar and it now feels a bit more intuitive. It is available in the latest nightly build.

daxliniere, 2022-04-13:
You're an absolute legend, Joacim!

daxliniere, 2022-04-20:
The bold filename text actually works really well, too. Would be cool to see an asterisk appended, but between the bold and the save button, this is a big improvement. :)

closed 2022-04-20 by daxliniere