winder/Universal-G-Code-Sender #1854: Permanent Editor pane

daxliniere, 2022-04-03:
### Description

Hey team,
Is it possible to keep this layout shown below? When there is no NC file loaded (i.e. on startup), the Editor pane is not visible and when you load a file, it goes to a default position of a second tab in the Visualizer's current pane. I was thinking, is it possible to create a dummy mode or something?

Also, there is an unused pane in the middle column, below the Visualizer and above the Jog Controller, that I can't seem to be able to remove.

[image]

All the best!
Dax.

risototh, 2023-04-07:
Would be happy too, because in my case it opens the editor in the smallest pane-group I have - right over the DRO :) I think, I just have to learn Java a bit...

breiler, 2023-04-09:
The window positioning is not optimal and there isn't much documentation on it. Depending on how messed up it has gotten you can use the menu option "Windows -> Reset windows" to restore to the defaults.

By default the layout should look something like this and consist of five default containers. Mostly you can drag windows around and place them where you like. New containers might get created depending how you dock the windows. And the windows should remember their positions. 

All except for the "Editor"-container. This houses the windows for any gcode or design file you open (along with the Welcome page). The editor will not remember its position if you try to move it. Any new file will load into the Editor container again. And unlike the other containers, the "Editor" container can never be removed, instead a gray empty area will be shown instead.

This is a NetBeans-specific thing and can't be changed at the moment.

[image]

risototh, 2023-04-09:
I know, that the Netbeans is a bit meesy in this - I used it a couple of years before Intellij PhpStorm... I never thought about it as a fixed layout on your image, now it makes sense, as I have three columns, but in the middle, I have DRO in the upper half and status in the bottom half. The Editor in my case is on the right upper area, along with the workflow and visualizer... So the best option how to "fix" this, is to swap the left and middle panes and just use it like that. Thanks for explanation.

daxliniere, 2023-04-09:
"Depending on how messed up it has gotten you can use the menu option "Windows -> Reset windows" to restore to the defaults."
Yes, but it doesn't default to my desired layout. :)

"This is a NetBeans-specific thing and can't be changed at the moment."
Man, NetBeans sucks. I'm sorry you've had to deal with so many of its limitations. <3

risototh, 2023-04-09:
I can confirm, that the Reset windows does it wrong even in the normal Netbeans :D