winder/Universal-G-Code-Sender #1862: Macros not updated until UGS is closed & reopened

daxliniere, 2022-04-21:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

I'm not Gcode expert, but argh! I've been pulling my hair out wondering why the code I've been issuing hasn't worked the way I thought and it's lead me to discovering a bug.

UGS doesn't update macros when you click OK or even if you click Apply first. (Generally OK = Apply + Close.)

### Operating System

Win10x64 Pro N

breiler, 2022-04-22:
Can't reproduce the problem. If you hit OK this is the same as Apply + Close. If you hit Apply the settings are saved, and you can close or hit cancel but the settings are already stored.

When I change a macro the settings are effective right away, tried different scenarios here:
[image]

daxliniere, 2022-04-22:
Hmm.. I don't know what to tell you, then. Do you think it makes any difference that I was using a multi-line macro?

```G21;G53 G90 Z-1 F500;G53 G90 X302 F1800;G53 G90 Z-25 F500;G91 G38.2 Z-46 F100;G0 Z1;G38.2 Z-2 F50;G10 L20 P1 Z4.985;G10 L20 P2 Z4.985;G10 L20 P3 Z4.985;G10 L20 P4 Z4.985;G10 L20 P5 Z4.985;G10 L20 P6 Z4.985;G90 G0 Z20;G90;```

daxliniere, 2022-04-22:
Well, I don't know what to say. I just crashed my Z-axis because the macro wasn't updated. :(

daxliniere, 2022-04-22:
Ahh, wait a moment.. I think that this time I might have clicked the Toolbox button instead of the Macro box button. The crash would be explained by this: https://github.com/winder/Universal-G-Code-Sender/issues/1827

daxliniere, 2022-06-07:
New information:
I can now confirm the specifics of this bug. Steps to reproduce:
1) Change a macro that is assigned to a Toolbox button,
2) Close Settings window,
3) Press the macro button in the Macros pane - the updated macro will be sent
4) Press the macro button in the Toolbox pane - the OLD macro will be sent.

PR #1959 (2022-08-11): Fixed macro shortcuts
Added a UUID to macros in order to identify them uniquely. Previously the name was used to register the macro action which is then stored with that file name. The name may contain illegal characters for the file system which would break any macro shortcut. 

The name of the macro action was not updated which is now also fixed.

Fixes #1901 
Fixes #1862
Fixes #1827
closed 2022-08-11 by breiler