winder/Universal-G-Code-Sender #1865: Download section not working

StefanM135, 2022-04-25:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

The download section on the github pages seems to be down. No download possible, only ending up on jfrog website with "unhealthy service at Frontend".

Kindly regards Stefan

breiler, 2022-04-25:
@winder is trying to make contact with the service provider and figure out whats wrong.

In the meantime I have temporarily uploaded the binaries under the release page:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.11

StefanM135, 2022-04-25:
> @winder is trying to make contact with the service provider and figure out whats wrong.
> 
> In the meantime I have temporarily uploaded the binaries under the release page: https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.11

Thanks for the fast response and for uploading it there

closed 2022-04-25 by StefanM135