winder/Universal-G-Code-Sender #1881: Status Freezes in Run and java.lang.OutOfMemoryError

rykellogg, 2022-05-08:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

System will run for a minute or so after connection then freezes most functionality and throws error:
java.lang.OutOfMemoryError: unable to create native thread: possibly out of memory or process/resource limits reached.

### How to reproduce

1. Open program.
2. Connect to machine (x-carve 750 in my case)
3. Send several commands via jog controller or macros.
4. at some point, status will not return to idle after previous commands and error icon will appear in lower right corner.

### Operating System

Windows 11 Home

### Anything else

Happens every time.
Things I've tried:
- Update java to most recent version
- Increased memory allocation to 256 per this [link](https://github-wiki-see.page/m/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview#:~:text=Increase%20the%20memory%20allocated%20to%20UGS%20by%20navigating,or%20larger.%20Additional%20details%20can%20be%20found%20here?msclkid=e255fb22ce6011ecb055da08470d94e8).
- Restarted several times.
- Deleted and reinstalled application off fresh download.

May be noteworthy that the ugsplatform.exe file has this issue. The ugsplatform64.exe crashes completely within seconds or a minute after launch. Suspicious I may have a common memory allocation issue here.

rykellogg, 2022-05-08:
Updated memory to 1024 per this ticket: https://github.com/winder/Universal-G-Code-Sender/issues/1844
Tried with visualizer turned off per same link.
Still no luck and no obvious improvements.

Also tried running in Win 8 compatibility mode with no luck.

rykellogg, 2022-05-08:
**From system:**
Processor	11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz   2.80 GHz
Installed RAM	16.0 GB (15.8 GB usable)
System type	64-bit operating system, x64-based processor
Edition	Windows 11 Home
Version	21H2
Installed on	‎4/‎24/‎2022
OS build	22000.652
Experience	Windows Feature Experience Pack 1000.22000.652.0

**From UGS**
Product Version: Universal Gcode Platform 20220321
Java: 13.0.1; OpenJDK Server VM 13.0.1+9
Runtime: OpenJDK Runtime Environment 13.0.1+9
System: Windows 10 version 10.0 running on x86; Cp1252; en_US (ugsplatform)
User directory: C:\Users\rykel\.ugsplatform\2.0.11\dev
Cache directory: C:\Users\rykel\.ugsplatform\2.0.11\dev\var\cache

breiler, 2022-05-09:
Hi, I think the instructions [here](https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview#program-slows-down-and-send-freezes) is wrong for setting the -Xms, this will set the initial heap size. What you probably want is to set the maximum heap size like this: `-J-Xmx1024m`

rykellogg, 2022-05-09:
I performed that change without any improvement. Continuing to sift to see if I have any strange PC configs, and will report back if I find any. 
Any other ideas?

breiler, 2022-07-24:
There is now a 64-bit version available here under the nightly build that you could try: 
https://github.com/winder/Universal-G-Code-Sender#downloads

rykellogg, 2022-08-07:
Thank you. I've got a job for this weekend. I'll try it out.

mrmowgli, 2022-09-30:
Ok, this isn't specific to windows. I think it's an issue with the bundled JDK or a system issue. I get the same exact error on Linux, and the linux version comes bundled with 13.

```OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/alewis/apps/ugsplatform-linux/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
OpenJDK 64-Bit Server VM warning: INFO: os::commit_memory(0x00007f33e7a0f000, 16384, 0) failed; error='Not enough space' (errno=12)
#
# There is insufficient memory for the Java Runtime Environment to continue.
# Native memory allocation (mmap) failed to map 16384 bytes for committing reserved memory.
# An error report file with more information is saved as:
# /home/alewis/apps/ugsplatform-linux/bin/hs_err_pid14714.log
```

The overall crash info is too big, and I don't know that I can attach the file to this issue?
Here's an excerpt the end with some runtime info: 

```
VM Arguments:
jvm_args: -Djdk.home=/home/mowgli/ugsplatform-linux/jdk/jdk-13.0.1+9-jre -Dnetbeans.dirs=/home/mowgli/ugsplatform-linux/etc:/home/mowgli/ugsplatform-linux/extra:/home/mowgli/ugsplatform-linux/ugsplatform:/home/mowgli/ugsplatform-linux/bin:/home/mowgli/ugsplatform-linux/ide:/home/mowgli/ugsplatform-linux/java:/home/mowgli/ugsplatform-linux/platform: -Dnetbeans.home=/home/mowgli/ugsplatform-linux/platform -Xms64m -Xverify:none -Dsun.java2d.noddraw=true -Dsun.awt.noerasebackground=true -Dnetbeans.indexing.noFileRefresh=true -Dorg.openide.text.big.file.size=20 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/home/mowgli/.ugsplatform/2.0-SNAPSHOT/dev/var/log/heapdump.hprof 
java_command: org.netbeans.Main --cachedir  --userdir /home/mowgli/.ugsplatform/2.0-SNAPSHOT/dev --branding ugsplatform
java_class_path (initial): /home/mowgli/ugsplatform-linux/platform/lib/boot.jar:/home/mowgli/ugsplatform-linux/platform/lib/org-openide-modules.jar:/home/mowgli/ugsplatform-linux/platform/lib/org-openide-util.jar:/home/mowgli/ugsplatform-linux/platform/lib/org-openide-util-lookup.jar:/home/mowgli/ugsplatform-linux/platform/lib/org-openide-util-ui.jar
Launcher Type: SUN_STANDARD

[Global flags]
     intx CICompilerCount                          = 4                                         {product} {ergonomic}
     uint ConcGCThreads                            = 2                                         {product} {ergonomic}
     uint G1ConcRefinementThreads                  = 8                                         {product} {ergonomic}
   size_t G1HeapRegionSize                         = 1048576                                   {product} {ergonomic}
    uintx GCDrainStackTargetSize                   = 64                                        {product} {ergonomic}
     bool HeapDumpOnOutOfMemoryError               = true                                   {manageable} {command line}
    ccstr HeapDumpPath                             = /home/mowgli/.ugsplatform/2.0-SNAPSHOT/dev/var/log/heapdump.hprof         {manageable} {command line}
   size_t InitialHeapSize                          = 67108864                                  {product} {command line}
   size_t MarkStackSize                            = 4194304                                   {product} {ergonomic}
   size_t MaxHeapSize                              = 8361345024                                {product} {ergonomic}
   size_t MaxNewSize                               = 5016387584                                {product} {ergonomic}
   size_t MinHeapDeltaBytes                        = 1048576                                   {product} {ergonomic}
   size_t MinHeapSize                              = 67108864                                  {product} {command line}
    uintx NonNMethodCodeHeapSize                   = 5836300                                {pd product} {ergonomic}
    uintx NonProfiledCodeHeapSize                  = 122910970                              {pd product} {ergonomic}
    uintx ProfiledCodeHeapSize                     = 122910970                              {pd product} {ergonomic}
    uintx ReservedCodeCacheSize                    = 251658240                              {pd product} {ergonomic}
     bool SegmentedCodeCache                       = true                                      {product} {ergonomic}
   size_t SoftMaxHeapSize                          = 8361345024                             {manageable} {ergonomic}
     bool UseCompressedClassPointers               = true                                 {lp64_product} {ergonomic}
     bool UseCompressedOops                        = true                                 {lp64_product} {ergonomic}
     bool UseG1GC                                  = true                                      {product} {ergonomic}

Logging:
Log output configuration:
 #0: stdout all=warning uptime,level,tags
 #1: stderr all=off uptime,level,tags

Environment Variables:
[... cut ...] ves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities

Memory: 4k page, physical 32659952k(11483976k free), swap 78785528k(78785528k free)

vm_info: OpenJDK 64-Bit Server VM (13.0.1+9) for linux-amd64 JRE (13.0.1+9), built on Oct 26 2019 10:07:46 by "jenkins" with gcc 7.3.1 20180303 (Red Hat 7.3.1-5)

END.
```

Bundled JDK version:
ugsplatform-linux/jdk/jdk-13.0.1+9-jre/

System Java version:
/usr/lib/jvm/java-11-openjdk-11.0.16.1.1-1.fc35.x86_64/

mrmowgli, 2022-09-30:
[hs_err_pid14714.log](https://github.com/winder/Universal-G-Code-Sender/files/9682347/hs_err_pid14714.log)
Here's the full crash file, which should be entertaining.

breiler, 2022-09-30:
Thanks for your detailed report @mrmowgli. The problem with windows is that it was using a 32-bit binary. A 32-bit environment only allows 2Gb to be allocated, for which the JVM will default use maximum 1/4th of which will be a bit low when loading a large model.

If I can decipher the log correctly your JVM instance is configured to use ~8Gb max heap space (MaxHeapSize = 8361345024) of a total of 31Gb. So that should be more than enough to run UGS unless you are loading a gigantic Gcode-files or there is some sort of memory leak.

I can see ~4000 references to `anon_inode:i915.gem` which I guess is graphics card related. Just out of curiosity, can you post your latest log file from: `~/.ugsplatform/2.X/dev/var/log/messages.log`

We have had reports of incompatible graphics drivers causing the program too spew errors to the log. (https://groups.google.com/g/universal-gcode-sender/c/pEz1tAZBrH4)

mrmowgli, 2022-10-01:
Ah fun times, ok looks like there is at least an actual exception. This is a dual video card laptop, perhaps it is trying to use the Intel driver, but GL is running on the Nvidia card. Hmm. Not sure how to force this from the command line.

```
        at java.desktop/javax.swing.JComponent.paint(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintToOffscreen(Unknown Source)
        at java.desktop/javax.swing.BufferStrategyPaintManager.paint(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paint(Unknown Source)
        at java.desktop/javax.swing.JComponent._paintImmediately(Unknown Source)
        at java.desktop/javax.swing.JComponent.paintImmediately(Unknown Source)
        at java.desktop/javax.swing.RepaintManager$4.run(Unknown Source)
INFO [null]: Last record repeated again.
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
INFO [null]: Last record repeated again.
        at com.jogamp.opengl.util.AWTAnimatorImpl$1.run(AWTAnimatorImpl.java:170)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
INFO [null]: Last record repeated again.
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
        at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:136)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
INFO [null]: Last record repeated again.
        at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
Caused by: com.jogamp.opengl.GLException: Profile GL4bc is not available on X11GraphicsDevice[type .x11, connection :1, unitID 0, handle 0x7f6965169b50, owner true, ResourceToolkitLock[obj 0x3c3a183f, isOwner true, <35ac425, 658f9d2b>[count 1, qsz 0, owner <AWT-EventQueue-0>]]], but: [GLProfile[GLES1/GLES1.hw], GLProfile[GLES2/GLES3.hw], GLProfile[GL2ES1/GLES1.hw], GLProfile[GL4ES3/GL4.hw], GLProfile[GL2ES2/GL4.hw], GLProfile[GL4/GL4.hw], GLProfile[GLES3/GLES3.hw], GLProfile[GL4/GL4.hw], GLProfile[GL3/GL4.hw], GLProfile[GL2GL3/GL4.hw]]
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:991)
        at jogamp.opengl.GLContextImpl.verifyInstance(GLContextImpl.java:1471)
        at jogamp.opengl.GLContextImpl.setGLFunctionAvailability(GLContextImpl.java:1942)
        at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:395)
        at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
        at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.initialize(GLJPanel.java:1731)
        ... 57 more
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Saving settings.
INFO [org.netbeans.ui.indexing]: Indexing started, time from last indexing 0 ms.
INFO [org.netbeans.ui.indexing]: Indexing finished, indexing took 4 ms.
```

mrmowgli, 2022-10-01:
Ok, so I set Nvidia as the primary driver and everything seems to be ok, no issues with memory or threads. Perhaps this should be mentioned somewhere in the docs?

breiler, 2022-10-01:
@mrmowgli excellent, feel free to write a couple of lines how you solved it here:
https://github.com/winder/Universal-G-Code-Sender/wiki/Troubleshooting-Overview

breiler, 2022-12-20:
Closing as this has been fixed for Windows with the 64-build. Linux will now work better as of 2.0.13 after our update of the JOGL-library.

closed 2022-12-20 by breiler