winder/Universal-G-Code-Sender #1891: program still running when $4=1

kweilim98, 2022-05-29:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

when the $4=1, mean all the axis motor already been enable, but the program still running and non stop 

### Operating System

cnc grbl

breiler, 2022-05-29:
I don't understand what you mean, please elaborate.

kweilim98, 2022-05-29:
when $4 is enable , mean all the axis motor will be enable when detect the current cut off when g-code is running. but the g-code program still running and not hold

kweilim98, 2022-05-29:
because I am using the door switch and connected to a relay. When the door is open, all axis motor and spindle will be enabled, but the g-code program not freeze

breiler, 2022-05-29:
The $4 setting is just for inverting the step enable pin and to my knowledge does not have anything to do with the door state.

But you might be right that UGS doesn't recognise the door state from the controller and will keep streaming commands. This requires some experimentation.

kweilim98, 2022-05-29:
Will UGS focus on this feature to stop the command continues streaming? Because I already try on that and found that have problem on there. Hope the author can upgrade or add-on on there. thx u.

breiler, 2023-02-23:
Just tested this, UGS recognizes/reads the door switch, enters a HOLD state and pauses any running gcode program.
Closing as I can't reproduce the error.

closed 2023-02-23 by breiler