winder/Universal-G-Code-Sender #1893: UGS close/crash no error

Bagenberry, 2022-05-30:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

Running a cut anywhere from  5 min to 1 hour, UGS will just close/exit without warning, without error. Router stops moving but spindle continues to spin. This has happened using many different files generated by Carbide Create.  I’ve had to turn off and unplug my cnc to reset as well as reboot my pc. I’ve gotten to the point I DON’T run any other programs while UGS is running, but that did not resolve my issue. I’m not a very pc oriented/technical guy, so any suggestions need to be worded as simple and straightforward as possible. 😬 I’m currently working on making a Catan board (many pieces and play area borders), and I’m just frustrated. I like UGS and would like to continue to use it, but I will tty Candle for the moment.  Thank you for reading, and any assistance would be greatly appreciated.

 Bagenberry

### How to reproduce

Let me know what I need to capture/save and submit the next time it happens and I’ll be more than happy to do so!!

### Operating System

Windows 10 Home Ver. 21H2 OS build 1904.1706, running x86; Cp1252

### Anything else

Carbide Create Build 648, UGS platform 20220321, Java 13.0.1, Open JDK Client VM 13.0.1+9, Open JDK Runtime VM 13.0.1+9

There is no pattern I’ve seen regarding which file I’m cutting or how long before it happens.

breiler, 2022-05-30:
Next time this happens, open a file explorer and locate the directory `C:/Users/[your username>]/.ugsplatform/2.0.11/var/log/messages.log` and create a new comment here and attach the log file (by dragging and dropping it). It will hopefully contain some clues to what is happening. 

Do you know how much ram memory your computer have?

Bagenberry, 2022-06-01:
Here are the files from the latest execution. 

Also, 8.0 GB ram (7.69 GB usable)

Thank you.

From: Joacim Breiler
Sent: Monday, May 30, 2022 12:05 PM
To: winder/Universal-G-Code-Sender
Cc: Bagenberry; Author
Subject: Re: [winder/Universal-G-Code-Sender] UGS close/crash no error (Issue #1893)

Next time this happens, open a file explorer and locate the directory C:/Users/[your username>]/.ugsplatform/2.0.11/var/log/messages.log and create a new comment here and attach the log file (by dragging and dropping it). It will hopefully contain some clues to what is happening.
Do you know how much ram memory your computer have?

Bagenberry, 2022-06-03:
[messages(sent).log](https://github.com/winder/Universal-G-Code-Sender/files/8835635/messages.sent.log)

I hope I did this right. ;-)

Bagenberry, 2022-06-03:
I had renamed it before I started using the software again. :-)

breiler, 2022-06-27:
Sorry for the late reply, that log file did not contain a normal user session, did you upload the latest log file?

I was expecting it contain log rows for when you opened the file and started sending it. But this log file only contained log rows of you started the software.

closed 2022-07-24 by breiler