winder/Universal-G-Code-Sender #1901: Macros starting with "->" cause disk path error

daxliniere, 2022-06-04:
### Version

2.0.11

### Hardware / Firmware

FluidNC

### What happened

To aid in visual reference, I changed my macro names from **Go to X0,Y0** to **-> X0,Y0** and in doing so, caused UGS to show an error for each macro.

daxliniere, 2022-06-05:
I can confirm that starting it with the ">" character also has problems.

breiler, 2022-06-07:
When creating a Macro it will also create an action in the Netbeans platform which are stored as files. On Windows the file name is limited to characters described here: https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file
```
The following reserved characters:
< (less than)
> (greater than)
: (colon)
" (double quote)
/ (forward slash)
\ (backslash)
| (vertical bar or pipe)
? (question mark)
* (asterisk)
```

We have to look at how the action name is stored. There might be a possibility to override it in the .nbattrs-file. This could also solve the problem with keyboard shortcut mapping when a macro is renamed.

daxliniere, 2022-06-07:
Thanks Joacim. I would have suggested this was a pretty low priority, but know that it might be connecged to the macro rename issue, it seems more pressing. Especially considering how it affects Toolbox buttons. (https://github.com/winder/Universal-G-Code-Sender/issues/1862#issuecomment-1148965409)

PR #1959 (2022-08-11): Fixed macro shortcuts
Added a UUID to macros in order to identify them uniquely. Previously the name was used to register the macro action which is then stored with that file name. The name may contain illegal characters for the file system which would break any macro shortcut. 

The name of the macro action was not updated which is now also fixed.

Fixes #1901 
Fixes #1862
Fixes #1827
closed 2022-08-11 by breiler