winder/Universal-G-Code-Sender #1905: GCode file starts loading, causes the program to hang

tillig, 2022-06-20:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

I attempted to load a GCode file generated by Fusion 360 for a CNC router. The file is ~28MB. I select the file to load, and the UI becomes unresponsive. The "folder" icon remains highlighted as though I'd clicked it but no other buttons can be clicked. Even after leaving the UI alone for 15+ minutes, it never completes loading and never reports an error.

I thought it was a memory issue, so I edited the `ugsplatform.conf` to have `-J-Xms512m` to give it 512MB of memory. I don't see any indication that it's running out of memory, but I also don't see the file ever finish loading. There are no messages in the log that indicate what's going on.

I have tried this on two different machines - Windows 11 on a Surface Book (Intel i7 2GHz) and Raspbian 10 Buster on a Raspberry Pi 4. Both display the same result.

[image]

[image]

### How to reproduce

[Here is the file with the issue.](https://github.com/winder/Universal-G-Code-Sender/files/8936791/file-with-issue.zip) To reproduce the issue, open it from UGS like you want to send it to your CNC machine. That's it.

### Operating System

Windows 11 Pro; Raspbian 10 Buster

### Anything else

I'll include the tail of the `messages.log` from the Raspberry Pi since that's my main sender for the CNC router I have but the Windows log looks identical. I see the `Setting gcode file` message, it "preprocesses" and finishes that, then says `About to process {}` and the log ends with mouse move messages. No errors or additional detail show up in the log after this.

<details>
<summary>messages.log - Raspberry Pi</summary>

```text
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Setting gcode file.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Start preprocessing
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Preprocessing /home/pi/share/2 - 2 End Mill - Finishing.nc to /tmp/1655685667233-0/2 - 2 End Mill - Finishing.nc_ugs_1655685667233
INFO [com.willwinder.universalgcodesender.utils.KeepAwakeUtils]: Moving the mouse location slightly to keep the computer awake.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Took 166957ms to preprocess
INFO [com.willwinder.ugs.nbm.visualizer.renderables.GcodeModel]: About to process {}
INFO [com.willwinder.universalgcodesender.utils.KeepAwakeUtils]: Moving the mouse location slightly to keep the computer awake.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated more than 10 times, further logs of this record are ignored until the log record changes.
```

</details>

I did go through issues to see if there was anything else I could tweak, but I couldn't find anything. I don't have the "Arc Expander" enabled, not sure if that matters.

I tried another fairly large (~26MB) file and that one, too, totally hangs the UI on both Windows and Raspberry Pi. I am able to load files that are smaller (~5-10MB) without issue. This makes it seem like it's file size related (or, I guess, more accurately, number of instructions).

Probably unrelated but interesting - on Windows, when running `ugsplatform64.exe`, it still appears to be running under a 32-bit JDK.

[image]

tillig, 2022-06-20:
I've been fighting this for hours, but I just now cranked the memory up to 1024 (1GB) and _now_ on the Raspberry Pi version, the file loads after... 10-ish minutes? And on Windows I get a prompt to ask me if I really want to open the file because it's too large and the UI will become unusable. That probably would have been a good prompt to see at lower memory levels. 😆 (If I do allow it to finish loading, sure enough, I do get the out of memory error.)

On Windows, I don't seem to be able to specify larger than about `1500m` for memory - I try to open UGS and after the splash screen it shuts down. At 1500 I still run out of memory.

Any way to figure out what's eating all the memory and get rid of it? If it's the visualizer, can I somehow disable that and free up the memory?

breiler, 2022-06-21:
Thanks for the detailed report!

I have no problems opening this file on MacOSX on a fairly old machine though it takes about 15 seconds for it to be parsed. The difference is that it uses a 64-bit jdk and I can see that it uses almost 4Gb of ram (which is alot)...

We are using the 32-bit JRE/JDK on Windows to make it easier for users to not bother which operating system they are using. We had people still running 32-bit Windows XP.

* You can try and disable the editor, it ate about 1Gb of ram: https://github.com/winder/Universal-G-Code-Sender/wiki/Usage#editor-settings
* You can also close the visualizer it ate about 2Gb of ram.
* You could try and download a  64-bit JDK on your Windows machine and change the path in `ugsplatform.conf` and see if that works better:
https://adoptopenjdk.net/releases.html?variant=openjdk13&jvmVariant=hotspot
https://github.com/AdoptOpenJDK/openjdk13-binaries/releases/download/jdk-13.0.2%2B8/OpenJDK13U-jre_x64_windows_hotspot_13.0.2_8.zip

tillig, 2022-06-21:
Cool, I'll give these tips a shot. It might be a couple of days before I can (I usually get weekends to play on the CNC). I assume the "Show editor when opening g-code files" is the setting you're talking about with "disable the editor." I can't remember if I tried opening the file without the visualizer tab open. If it's not open, does that skip the visualizer calculations/processing? And, again, thanks for the tips.

tillig, 2022-06-26:
Turns out I'd already had the editor disabled, but closing the visualizer window at least lets the file load on the Raspberry Pi, which is honestly my primary use case - I don't have a laptop I can leave connected to the CNC while it cuts for hours.

I do see there's an exception in the log during loading, unsure if that's of consequence:

```text
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Removing UGSEvent listener: RendererInputHandler
INFO [com.willwinder.ugs.nbm.visualizer.Visualizer2TopComponent]: Component closed, panel = AWT-GLJPanel[ drawableType null, chosenCaps null]
WARNING [org.openide.filesystems.Ordering]: Not all children in OptionsDialog/ marked with the position attribute: [UGS.instance], but some are: [Actions, Advanced, Appearance, Editor, Keywords, PreviewExamples, UGS, General.instance, Editor.instance, FontsAndColors.instance, Keymaps.instance, Appearance.instance, Advanced.instance]
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Setting gcode file.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Start preprocessing
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Preprocessing /home/pi/share/MACUSA Before Smoothing/2 - 2 End Mill - Finishing.nc to /tmp/1656266331447-0/2 - 2 End Mill - Finishing.nc_ugs_1656266331448
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Took 260572ms to preprocess
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Getting processed gcode file.
INFO [null]: Last record repeated again.
WARNING [org.netbeans.core.TimableEventQueue]: too much time in AWT thread null
SEVERE [global]
java.lang.NullPointerException
	at com.willwinder.ugs.nbp.core.actions.OpenAction.actionPerformed(OpenAction.java:95)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:290)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:136)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
INFO [org.netbeans.core.ui.focus]: External Changes Refresh on focus gain disabled
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: OutlineAction
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Getting processed gcode file.
INFO [null]: Last record repeated again.
```

Once it was loaded, I tried opening the visualizer to see what would happen. It took a while, the visualizer window opened, but it never actually showed the rendered preview - it was only the grid with the position of the CNC router bit. I'm guessing it ran out of memory and gave up. I did not see anything in the log about it. It output a message "Initializing OpenGL context" and that was it.

Anyway, it sounds like turning off the visualizer is the answer for me in these cases.

I wonder if it would be worth adding something about disabling the editor and/or visualizer [on the Troubleshooting page](http://winder.github.io/ugs_website/guide/troubleshooting/) at the bottom where it talks about increasing the available memory.

closed 2022-06-26 by tillig