winder/Universal-G-Code-Sender #1908: UGS 2.0.11 keeps on crashing

drnathwani1, 2022-06-22:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

Tring to Install UGS 2.0.11 64 version in Windows 11 Home 64 bit Version 10.0..22000 BUILD 22000 . After EGS opens it crashes in 10-15 seconds. Tried multiple time, doing same thing.

### Operating System

Windows 11 64 bit

### Anything else

everytime, after opening UGS, it crashes in 10 seconds
[hs_err_pid15904.log](https://github.com/winder/Universal-G-Code-Sender/files/8960822/hs_err_pid15904.log)
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/8960823/messages.log)
.

breiler, 2022-06-23:
It is hard to tell from the logs, but the problem seems to be related to OpenGL. We have seen several people having problems with drivers for their graphics cards. Once they upgraded/downgraded it started working. I don't think there is much we can do about this problem in UGS unfortunately.

drnathwani1, 2022-06-23:
No problem, thank you. I will have to look around for other softwares.

EW-N3UE, 2022-06-24:
I have the same problem.
New HP notebook Win 11 Home 64bit
Drivers all updated. did the MS one, then tried the HP one.
Did notice even if I unplug  the USB cable, still starts up and in a few seconds quits.
Also going to look for different software...

drnathwani1, 2022-06-25:
I downloaded gsender software,  it has been installed fine, let us see.

EW-N3UE, 2022-06-25:
I've downloaded OpenBuilds Control, but just checked gsender since you mentioned it... Yeah... Have to see...
At least they stay running!

closed 2022-06-27 by breiler