winder/Universal-G-Code-Sender #1910: Main menu and Drop-down menus are not in English

LeeRossCritchlow, 2022-06-24:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

Main menu and Drop-down menus are not in English.

### How to reproduce

Downloaded  from GitHub.

### Operating System

Windows 10

### Anything else

Every time

breiler, 2022-06-27:
It will attempt to load a language that your operating system is configured for. If this didn't work for some reason you can change this setting by opening the preferences:

[image]

LeeRossCritchlow, 2022-06-27:
I need to navigate to the UGS page. Please find attached a screen capture showing the language
[image]
.

LeeRossCritchlow, 2022-07-02:
I  am not sure how this got marked as  completed.
By clicking on menu and drop-down items I was able to get to the UGS page. The language was already English, I clicked Apply but the menu and drop-downs did not change.
I tried the ugsplatform.exe and ugsplatform64.exe, neither came up in English. I downloaded and replaced from a different vendor web page, still not coming up in English.
I am nearing completion of the construction of my X-Y-Z setup. I need to be able to use this program.
Please advise.

breiler, 2022-07-02:
Try clearing all settings:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#user-content-configuration-files

If that doesn't work attach your log file from `C:/Users/[your username>]/.ugsplatform/2.X/dev/var/log/messages.log`

LeeRossCritchlow, 2022-07-02:
Joacim,

Renamed UniversalGcodeSender.json,  firmware_config; did not  solve problem.

Please  find attached the  log file.

 

https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#user-content-configuration-files

breiler, 2022-07-02:
You can't attach files by replying to email. Please add the log file from github web page.

LeeRossCritchlow, 2022-07-03:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9033899/messages.log)

LeeRossCritchlow, 2022-07-03:
Should  be there  now.

 �

 �

breiler, 2022-07-03:
The log did not give any clues. 
Try deleting or renaming your settings directory: C:\Users\Critc\.ugsplatform\2.0.11\

LeeRossCritchlow, 2022-07-03:
Joacim,

Renamed Dev folder, still not English.

Do you recognize the language? Could it be a Unicode mismatch?

I have checked Windows Display language, everything is English.

Could you send me snapshots of English Menus, Pull-downs, and ToolTips so I can begin to use the program (or possibly source code that  I can debug in Visual Studio)?

Thanks,

lee

 �

 �

breiler, 2022-07-04:
Someone else have had the same problem:
https://stackoverflow.com/questions/66714335/apache-netbeans-12-0-or-12-3-window-tabs-label-text-language-issue

Open the file `ugsplatform/etc/ugsplatform.conf` and try to add `-J-Dfile.encoding=UTF-8` to the `default_options`.

LeeRossCritchlow, 2022-07-04:
Joacim,

I cannot find ‘jdkhome="jdk/jdk-13.0.1+9-jre"’ as specified in ugsplatform.conf.

I installed Java, no change to menu.

I download the latest JDK (18).

I cannot seem to find the correct syntax for jdkhome. I get the attached pop-up. The program loads modules then dies.

Please advise.

 �

 �

https://stackoverflow.com/questions/66714335/apache-netbeans-12-0-or-12-3-window-tabs-label-text-language-issue

breiler, 2022-07-04:
You still can't attach files (or images) through email, in that case you need to answer on [github](https://github.com/winder/Universal-G-Code-Sender/issues/1910).

I never wrote that you should update JDK, I asked you to update the line with `default_options` with the parameter that says that files should be opened using UTF-8 (`-J-Dfile.encoding=UTF-8`), like the following: 
```
# ${HOME} will be replaced by user home directory according to platform
default_userdir="${HOME}/.${APPNAME}/2.0-SNAPSHOT/dev"
default_mac_userdir="${HOME}/Library/Application Support/${APPNAME}/2.0-SNAPSHOT/dev"

# options used by the launcher by default, can be overridden by explicit
# command line switches
default_options="-J-Dfile.encoding=UTF-8 --branding ugsplatform -J-Xms64m -J-Xverify:none -J-Dsun.java2d.noddraw=true -J-Dsun.awt.noerasebackground=true -J-Dnetbeans.indexing.noFileRefresh=true -J-Dorg.openide.text.big.file.size=20"
# for development purposes you may wish to append: -J-Dnetbeans.logger.console=true -J-ea

# default location of JDK/JRE, can be overridden by using --jdkhome <dir> switch
jdkhome="jdk/jdk-13.0.1+9-jre"

# clusters' paths separated by path.separator (semicolon on Windows, colon on Unices)
#extra_clusters=
```

If you tried that and it still doesn't work you could try and update to a later JDK. However UGS only works properly across multiple platforms using the bundled Java 13.0.1+9 due to limitations in some of the libraries we are using. The latest Java 18 release will probably not work.

You could try JRE 13.0.2+8: https://adoptopenjdk.net/archive.html?variant=openjdk13&jvmVariant=hotspot
Unzip it to ugsplatform/jdk/ and point to it by setting the `jdkhome`-property in the configuration file.

LeeRossCritchlow, 2022-07-04:
I apologize for getting desperate.
Attached is the current conf file.
jdkhome="jdk/jdk-13.0.1+8-jre" did not solve the current the issue.
[ugsplatform.zip](https://github.com/winder/Universal-G-Code-Sender/files/9039264/ugsplatform.zip)

breiler, 2022-07-04:
What did you try?
You have set the `jdkhome="jdk/jdk-13.0.1+8-jre"`, did you dow

breiler, 2022-07-04:
Well, in your desperation you are leaving out details which makes it hard to help you.

I can see that your config now includes the `-J-Dfile.encoding=UTF-8`which looks right.
But your `jdkhome` is set to `jdk/jdk-13.0.1+8-jre`. We are shipping UGS with `jdkhome="jdk/jdk-13.0.1+9-jre"`. Did you download another JRE?

LeeRossCritchlow, 2022-07-04:
yes

 �

 �

LeeRossCritchlow, 2022-07-04:
Hey man, I  did it per your instructions 7 hours ago.

LeeRossCritchlow, 2022-07-04:
My bad, I changed the 9 to an 8 did not catch that the 1 changed to a 2.
Made the change, no change in the issue.

breiler, 2022-07-04:
Alright, too bad. =(

Unfortunately I can't reproduce the error and this is the first time I have heard of this problem. I don't think that there is much I can do about it.

There are plenty of other alternatives out there for you to try out:
https://github.com/winder/Universal-G-Code-Sender/wiki/Software-comparison

breiler, 2022-12-16:
Can't reproduce the error

closed 2022-12-16 by breiler