winder/Universal-G-Code-Sender #1911: Joystick: Analog jog modes' feed rate

daxliniere, 2022-06-27:
### Description

WHAT: I just started using a PS3 controller to jog my CNC mill and I'ver been wonder "where have you been all my life?!" 😆

The only issue I'm seeing so far is that we kinda need separate feed rates for X/Y and Z axes. UGS is set to 3000mm/sec, which is perfect for X/Y, but is definitely too fast for Z axis.

Adding separate feed rate value for Z axis would solve this.

HOW: Aside from the obvious adding another feed rate numerical field to the GUI (I have ABC step size field hidden, by the way), another solution might be to add a field or option to the Joystick page which reduces the Z feed rate by a factor of 10.

breiler, 2022-06-28:
It is cool right? 😀 

I think that in order to have separate speeds for XY and Z we would need add a limitation that you only can jog one plane at the time. For instance, you will be able to jog diagonally in XY but not in XZ because we can only set one feed rate at a time in gcode:
`
$J=G91G0X10Z10F1000
`

But that is maybe ok?

daxliniere, 2022-06-28:
Yes, I think that would be good and actually an added benefit, as it would avoid accidentally bumping XY/Z while adjusting Z/XY.