winder/Universal-G-Code-Sender #1915: Visualizer machine boundries in unexpected place

wohali, 2022-07-05:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

I've configured UGS and GRBL 1.1h.20190830 with the following:

```
$20=1 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=3 (Homing direction invert - home is at top left front on this machine)
$130=110.900 (X-axis maximum travel)
$131=65.000 (Y-axis maximum travel)
$132=53.000 (Z-axis maximum travel)
```

After loading some GCode, homing the machine, and then a Reset Zero (`G10 P0 L20 X0 Y0 Z0`), I have coordinates as expected via `$#`:

```
[G54:-107.900,-62.000,-3.000]
```

But the build volume in the visualizer is still way off:

[image]

I can adjust the work position to place the design within the bounding box, but the machine head is then outside of the volume - and all motion is disabled because of soft limits. For example, moving the work to the very edge of the volume results in this (note the `$#` output for G54): 

[image]

I looked at #1287 for history on this issue, but no help. Something seems amiss in the positioning of the build volume for negative offsets.

### How to reproduce

Can be reproduced using the data given above for a machine and any g-code file.

### Operating System

Debian linux bullseye x86_64

breiler, 2022-07-06:
Thanks for the issue report, it got inverted in #1287 which might not be correct (it currently doesn't work on my machine either). I am not sure, but I think @rlwoodjr uses `HOMING_FORCE_ORIGIN` to make the machine work in positive space.

So to fix this we would need a manual setting in the visualiser to invert the bounding box and I just haven't gotten around to it yet.

wohali, 2022-07-06:
Thanks for the info. I never considered `HOMING_FORCE_ORIGIN` since that'd make this mill different from every other mill I've used 😉 Makes sense, though.

PR #1964 (2022-08-17): Visualizer settings
Now stores which renderables that should be active and added those settings to the visualizer options.
Also added soft limits for FluidNC with the possibility to invert the machine boundaries.
[image]

Fixes: #1915
Fixes: #1691
closed 2022-08-17 by breiler