winder/Universal-G-Code-Sender #1917: Can not parse grbl file

weizhuowang, 2022-07-08:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

I generated this gcode from fusion 360 using grbl post processor. As I try to visualize it with the UGS platform it gave me this error.
[image]

### Operating System

Windows 10

breiler, 2022-07-08:
Hmm, odd. Try and remove the first row with a comment in the file `(ff)`.

I know that I have a fix for the parsing a file where the first line is empty in the next of UGS. This might be related with that problem.

weizhuowang, 2022-07-08:
Yeah I also tried to remove all the comments, still the same error. 
[image]

If I use the classic version I get this error output from the command line:
```
Jul 08, 2022 2:30:53 PM com.willwinder.universalgcodesender.model.GUIBackend setGcodeFile
INFO: Setting gcode file.
Jul 08, 2022 2:30:53 PM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFO: Start preprocessing
Jul 08, 2022 2:30:53 PM com.willwinder.universalgcodesender.model.GUIBackend preprocessAndExportToFile
INFO: Preprocessing C:\Users\Wang Weizhuo\OneDrive - Stanford\Spring 2022\Misc\CNC\1001.nc to C:\Users\Wang Weizhuo\AppData\Local\Temp\1657308653819-0\1001.nc_ugs_1657308653820
Jul 08, 2022 2:30:53 PM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFO: Took 88ms to preprocess
Jul 08, 2022 2:30:53 PM com.willwinder.universalgcodesender.model.GUIBackend getProcessedGcodeFile
INFO: Getting processed gcode file.
Jul 08, 2022 2:30:53 PM com.willwinder.universalgcodesender.utils.SettingsFactory saveSettings
INFO: Saving settings.
Jul 08, 2022 2:30:57 PM com.willwinder.universalgcodesender.visualizer.VisualizerCanvas generateObject
SEVERE: Problem opening file : Corrupt data found while processing gcode stream: G90 G94??G90G94??1??
java.io.IOException: Corrupt data found while processing gcode stream: G90 G94??G90G94??1??
        at com.willwinder.universalgcodesender.utils.GcodeStreamReader.getNextCommand(GcodeStreamReader.java:95)
        at com.willwinder.universalgcodesender.visualizer.GcodeViewParse.toObjFromReader(GcodeViewParse.java:143)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.generateObject(VisualizerCanvas.java:476)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.setFile(VisualizerCanvas.java:199)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.setProcessedGcodeFile(VisualizerCanvas.java:183)
        at com.willwinder.universalgcodesender.visualizer.VisualizerWindow.setProcessedGcodeFile(VisualizerWindow.java:94)
        at com.willwinder.universalgcodesender.MainWindow.setVisualizerFile(MainWindow.java:1801)
        at com.willwinder.universalgcodesender.MainWindow.visualizeButtonActionPerformed(MainWindow.java:1199)
        at com.willwinder.universalgcodesender.MainWindow.access$2000(MainWindow.java:84)
        at com.willwinder.universalgcodesender.MainWindow$22.actionPerformed(MainWindow.java:870)
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
        at java.awt.Component.processMouseEvent(Unknown Source)
        at javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.awt.Component.processEvent(Unknown Source)
        at java.awt.Container.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

Jul 08, 2022 2:30:57 PM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: VisualizerWindow
Jul 08, 2022 2:30:57 PM com.willwinder.universalgcodesender.visualizer.VisualizerCanvas init
[... 7 lines cut ...]
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.init(VisualizerCanvas.java:221)
        at jogamp.opengl.GLDrawableHelper.init(GLDrawableHelper.java:644)
        at jogamp.opengl.GLDrawableHelper.init(GLDrawableHelper.java:667)
[... cut ...]
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

Jul 08, 2022 2:30:57 PM com.willwinder.universalgcodesender.visualizer.VisualizerCanvas reshape
INFO: Reshaping OpenGL context.
```

weizhuowang, 2022-07-09:
Interesting finding: I set my windows language to english and now it's parsing just fine. Maybe the language setting or encoding is somehow affecting the gcode parsing?

breiler, 2022-07-09:
Can you attach the file?
Did you use Chinese as language?

We have had a similar problem before where parsing number was a bit different in different locales. As an example in Sweden we use commas to separate decimals `0,401`. In some locales the minus character uses UTF-8 minus signs `0xE2 0x88 0x92` instead of the ASCII `0x45`.

These locale based settings for handling numbers are overridden by UGS but there might be other edge cases that we have missed.

weizhuowang, 2022-07-10:
Of course, check out the file below. I had to change the extension to .txt to upload it to github. Yes I used Chinese
[ff.nc.txt](https://github.com/winder/Universal-G-Code-Sender/files/9078306/ff.nc.txt)
.

breiler, 2022-12-15:
Fixed with #2082

closed 2022-12-15 by breiler