winder/Universal-G-Code-Sender #1924: UGS Screen Freeze and/or close out

chief4vfd, 2022-07-14:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

When running the 64 bit module, the program will terminate on its own without any warning or any error code. Sometimes this will happen when I first open the program, but it mainly happens while the project is running. When it does this, the spindle with continue for a brief minute and then stop. I did go into the ugsplatform.conf and modify the value of Xms from 64 to 512. This did not effect the issue that I am having. I did attempt to run the 32 bit and it will complete the project. However, the program would freeze during the project and all the buttons would be unresponsive. I would have to shut the program down. 

### How to reproduce

This issue easily reproduces on my computer. 

### Operating System

Windows 11 Home Version 21H2

breiler, 2022-07-15:
The `-Xms` only sets initial heap size to allocate, you would need to set instead `-Xmx` (I found the error on the troubleshooting page and corrected it). 

Currently we use a 32-bit JVM so we are limited to up to 4Gb of ram. But I have read that the 32-bit JVM is limited to only 2Gb, so to start with you could try to set it to `-Xmx=2048m`.

From you log I find this a bit strange, you are running the application from `C:\Users\melpo\OneDrive\Desktop\UGS V11\ugs-platform-app-win\ugsplatform-win` but it tries to load files from what I guess is an old location `C:\Users\melpo\Downloads\ugs-platform-app-win\ugsplatform-win\platform\modules\org-netbeans-core-ui.jar`. 
Odd that I haven't noticed this before, and it is easy to reproduce and could very well lead to some sort of memory leak.

Delete this cache directory `C:\Users\melpo\.ugsplatform\2.0.11\dev\var\cache` and those errors should go away.

We should try and find a way to programmatically invalidate the cache directory.

chief4vfd, 2022-07-15:
Thanks for the quick response!! So, I did find that I had 2.0.8 and 2.0.9 as well as 2 instances of 2.0.11 on the system. I did some house cleaning and now have a clean install of 2.0.11. I changed the memory allocation to -Xmx2048m and it will not open the program. A quick blink of the icon is all that occurs. So I changed the allocation to -Xmx1024m and now the program will open. However, it will close out once I start to manipulate any modules. I have attached the new message log for your review. I'm looking forward to your response.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9123352/messages.log)

breiler, 2022-07-15:
Hmm.

1. Delete or rename the settings directory: `C:\Users\melpo\.ugsplatform` to ensure there isn't any settings that mess things up.
2. Rename the application directory `C:\Users\melpo\ugs-platform-app-win (1)\` to `C:\Users\melpo\ugs-platform-app-win\` without spaces or parentheses.

chief4vfd, 2022-07-15:
I deleted the settings directory and renamed the application directory. No change. Still closes out when I start to manipulate any of the modules. New Message log attached.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9123608/messages.log)

chief4vfd, 2022-07-16:
I also found this error log under the c:\Users\melpo\ugs-platform-app-win\ugs-platform-app-win file.

[hs_err_pid22608.log](https://github.com/winder/Universal-G-Code-Sender/files/9125435/hs_err_pid22608.log)

breiler, 2022-07-17:
Really strange, I am wondering if this has to do with running a 32-bit JVM on 64-bit platform. I have not been able to reproduce this on my brand new laptop running Windows 10 though so I find it strange.

I'll try to create a new 64-bit build for windows for you to try.

chief4vfd, 2022-07-20:
Sorry for the delayed response. Thanks for all your assistance.

breiler, 2022-07-21:
There is a new 64-bit variant for windows here: https://ugs.jfrog.io/artifactory/UGS/nightly/ugs-platform-app-win64.zip

chief4vfd, 2022-07-23:
I downloaded the new 64-bit variant and was able to run the software without the program ever closing out. I also ran several projects that have always given me the issues such as closing out in the middle of the project as well as pausing the project during its run, which when restarted would usually miss some code and cut crazy. 

Thank you for you time and effort on resolving this issue.

breiler, 2022-07-24:
Thanks for your help!

closed 2022-07-24 by breiler

chief4vfd, 2022-10-23:
Good morning. I only have two things. First, UGS has been working flawlessly since your new 64-bit variant. My question now is that Java just popped up with an update. I wanted to check with you first before updating. Wasn't sure if I needed to remain on the version that I am currently running. Again, thanks for everything!! Mel

breiler, 2022-10-23:
UGS comes bundles with its own Java. So the Java-update will not effect UGS.

chief4vfd, 2022-10-23:
Thanks for the quick response. Just saw the Sponsor button, so that is the least I can do for you! Again, thanks for your time and assistance. Mel

breiler, 2022-10-23:
Whoho! You are my first sponsor! ❤️

chief4vfd, 2024-10-08:
Good morning. UGS has been working perfectly since the fix in 2022 (2022-7-22). I did want to ask if you thought if would be alright to upgrade to the latest version or stick with the version that I have been using since the fix. Thanks again, Mel.

breiler, 2024-10-08:
@chief4vfd I am glad to hear that it is working properly for you.

A lot has happened in UGS since that version with 266 commits (changes to ~2000 files) containing new features and bugfixes. 

If everything is working for you today and you aren't missing any features I don't see any point on upgrading. However, if you were to install a new system or if you encounter an annoying behavior or bug I would probably want to install the latest version.

fodderrlb, 2024-10-29:
I am having the same problem as above with UGS version 2.1.9 but I don't have the Xmx=64m line to change.  What I am working with is this: 
default_userdir="${DEFAULT_USERDIR_ROOT}"
default_cachedir="${DEFAULT_CACHEDIR_ROOT}/var/cache"

# options used by the launcher by default, can be overridden by explicit
# command line switches
default_options="--branding ugsplatform --clearcache -J-XX:MaxRAMPercentage=85.0 -J-Xverify:none -J-Dsun.java2d.noddraw=true -J-Dsun.awt.noerasebackground=true -J-Dpolyglot.engine.WarnInterpreterOnly=false -J-Dnetbeans.indexing.noFileRefresh=true -J-Dorg.openide.text.big.file.size=20 -J--add-opens=java.base/java.net=ALL-UNNAMED -J--add-opens=java.base/java.lang.ref=ALL-UNNAMED -J--add-opens=java.base/java.lang=ALL-UNNAMED -J--add-opens=java.base/java.security=ALL-UNNAMED -J--add-opens=java.base/java.util=ALL-UNNAMED -J--add-opens=java.base/java.nio=ALL-UNNAMED -J--add-opens=java.prefs/java.util.prefs=ALL-UNNAMED -J--add-opens=java.desktop/javax.swing.plaf.basic=ALL-UNNAMED -J--add-opens=java.desktop/javax.swing.text=ALL-UNNAMED -J--add-opens=java.desktop/javax.swing=ALL-UNNAMED -J--add-opens=java.desktop/java.awt=ALL-UNNAMED -J--add-opens=java.desktop/java.awt.event=ALL-UNNAMED -J--add-opens=java.desktop/sun.awt.X11=ALL-UNNAMED -J--add-opens=java.desktop/javax.swing.plaf.synth=ALL-UNNAMED -J--add-opens=java.desktop/com.sun.java.swing.plaf.gtk=ALL-UNNAMED -J--add-opens=java.desktop/sun.awt.shell=ALL-UNNAMED -J--add-opens=java.desktop/sun.awt.im=ALL-UNNAMED -J--add-exports=java.base/sun.reflect.annotation=ALL-UNNAMED -J--add-exports=java.desktop/sun.awt=ALL-UNNAMED -J--add-exports=java.desktop/java.awt.peer=ALL-UNNAMED -J--add-exports=java.desktop/com.sun.beans.editors=ALL-UNNAMED -J--add-exports=java.desktop/sun.swing=ALL-UNNAMED -J--add-exports=java.desktop/sun.awt.im=ALL-UNNAMED -J--add-exports=java.desktop/com.sun.java.swing.plaf.motif=ALL-UNNAMED -J-XX:+IgnoreUnrecognizedVMOptions"
# for development purposes you may wish to append: -J-Dnetbeans.logger.console=true -J-ea

# default location of JDK/JRE, can be overridden by using --jdkhome <dir> switch
jdkhome="jdk/jdk-17.0.8.1+1-jre"

# clusters' paths separated by path.separator (semicolon on Windows, colon on Unices)
#extra_clusters=

So I don't see a way to change the amount of memory there is just the XX:MaxRAMPercentage=85.0 line that has anything to do with memory.

Any help would be appreciated.

breiler, 2024-10-29:
@fodderrlb with the newer version of Java it is possible to use a percentage instead of a fixed memory limit. It will use at max 85% of the total system memory. If you have more than 2Gb of total system memory you should not need to change this setting.

What makes you think that UGS needs more memory?

fodderrlb, 2024-10-30:
When I try to load a file over 19.9mb it gives me an error that the file is too large.  If I try to run a file 20mb or larger it will stop running part way through.  The error I get is this:  The file mountain cabin1.gcode seems to be too large (42Mb) to safely open.  Opening the file could cause Out of Memory Error, which would make the IDE unusable.  Do you really want to open it.

breiler, 2024-10-30:
@fodderrlb it is not an error that it has run out of memory, just a warning that it could. If you have more than a couple of gigs of RAM on your system - it should be ok to press Yes on the warning dialog to continue opening the file. 

Running such large programs is however not tested by us, I would use caution running such large files.