winder/Universal-G-Code-Sender #1927: crash on quit (joystick thread)

bertieconfundo, 2022-07-16:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

When quitting UGS, MacOS reports a crash. Here is the stack trace of the crashed thread (looks to be related to joystick code):

Thread 1098 Crashed:: */Stop
0   libsystem_kernel.dylib        	    0x7ff80d03a00e __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff80d0701ff pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff80cfbbd24 abort + 123
3   libjvm.dylib                  	       0x1099c0fb8 os::abort(bool, void*, void const*) + 22
4   libjvm.dylib                  	       0x109b80a34 VMError::report_and_die(int, char const*, char const*, __va_list_tag*, Thread*, unsigned char*, void*, void*, char const*, int, unsigned long) + 2770
5   libjvm.dylib                  	       0x109b7ff3c VMError::report_and_die(Thread*, unsigned int, unsigned char*, void*, void*, char const*, ...) + 152
6   libjvm.dylib                  	       0x109b80a81 VMError::report_and_die(Thread*, unsigned int, unsigned char*, void*, void*) + 33
7   libjvm.dylib                  	       0x1099c489c JVM_handle_bsd_signal + 794
8   libjvm.dylib                  	       0x1099c23ee signalHandler(int, __siginfo*, void*) + 45
9   libsystem_platform.dylib      	    0x7ff80d085dfd _sigtramp + 29
10  ???                           	    0x70004b94c050 ???
11  CoreFoundation                	    0x7ff80d15796c CFRunLoopRemoveSource + 670
12  IOKit                         	    0x7ff80fa5f3d5 IOHIDDeviceUnscheduleFromRunLoop + 108
13  IOKit                         	    0x7ff80fa6366e __IOHIDManagerDeviceApplier + 556
14  CoreFoundation                	    0x7ff80d101766 __CFSetApplyFunction_block_invoke + 18
15  CoreFoundation                	    0x7ff80d1015d3 CFBasicHashApply + 115
16  CoreFoundation                	    0x7ff80d101523 CFSetApplyFunction + 131
17  IOKit                         	    0x7ff80fa62116 __ApplyToDevices + 105
18  IOKit                         	    0x7ff80fa62206 IOHIDManagerUnscheduleFromRunLoop + 75
19  IOKit                         	    0x7ff80fa62160 IOHIDManagerClose + 34
20  libjamepad64.dylib            	       0x15f96cbb2 DARWIN_JoystickQuit + 82
21  libjamepad64.dylib            	       0x15f8c2e85 SDL_JoystickQuit + 69
22  libjamepad64.dylib            	       0x15f89f22b SDL_QuitSubSystem_REAL + 155
23  libjamepad64.dylib            	       0x15f89f425 SDL_Quit_REAL + 21
24  ???                           	       0x10ffa6c90 ???
25  ???                           	       0x10ffa0fb3 ???
26  ???                           	       0x10ffa0fb3 ???
27  ???                           	       0x10ffa0ff8 ???
28  ???                           	       0x110dbc8bc ???
29  ???                           	       0x10ff97949 ???
30  libjvm.dylib                  	       0x10973ba89 JavaCalls::call_helper(JavaValue*, methodHandle const&, JavaCallArguments*, Thread*) + 619
31  libjvm.dylib                  	       0x10973ae7d JavaCalls::call_virtual(JavaValue*, Klass*, Symbol*, Symbol*, JavaCallArguments*, Thread*) + 231
32  libjvm.dylib                  	       0x10973af3a JavaCalls::call_virtual(JavaValue*, Handle, Klass*, Symbol*, Symbol*, Thread*) + 98
33  libjvm.dylib                  	       0x1097b84f6 thread_entry(JavaThread*, Thread*) + 120
34  libjvm.dylib                  	       0x109b2d8d4 JavaThread::thread_main_inner() + 134
35  libjvm.dylib                  	       0x109b2d712 JavaThread::run() + 292
36  libjvm.dylib                  	       0x109b2b4f3 Thread::call_run() + 113
37  libjvm.dylib                  	       0x1099c0871 thread_native_entry(Thread*) + 306
38  libsystem_pthread.dylib       	    0x7ff80d0704e1 _pthread_start + 125
39  libsystem_pthread.dylib       	    0x7ff80d06bf6b thread_start + 15

### How to reproduce

Run UGS (nightly builds have been doing this now for several months - might have been because I started using a joystick controller for jogging).  Connect to GRBL. Disconnect. Quit UGS (crashes).

### Operating System

MacOS v12.4 Monterey

### Anything else

I am going to look into what's going on here and see if I can submit a fix for this.

breiler, 2022-07-17:
I have been thinking of switching library from https://github.com/williamahartman/Jamepad to https://github.com/electronstudio/sdl2gdx as its specs looks a bit better (unsure about sdl2gdx on Linux ARM though). 

I'm not sure about the status of either projects anymore, if you have some time to look into it I would appreciate it!

bertieconfundo, 2022-07-19:
I don't have the time to look into changing over to a different API, but I will see about coming up with a fix for this crash. The really annoying aspect of this crash is that since it occurs on quit of UGS, it can sometimes crash UGS in the middle of writing out it's UI state and/or macros files - if that happens, one or both get corrupted which is really inconvenient.

bertieconfundo, 2022-07-23:
I was able to get everything working properly with Jamepad - no more crash on quit, and hot plug/unplug works now as well. (at least, on Mac it does... would be good to test it on other platforms)

closed 2022-08-08 by breiler