winder/Universal-G-Code-Sender #1929: re-opening the same file doesn't actually load the new gcode

bertieconfundo, 2022-07-16:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

if I have file.nc loaded, and then I go and update file.nc (post new gcode from Fusion 360, for example), and then try to open file.nc again in UGS, it doesn't actually re-load the changes from the file that were just written.  If I load a different file and then go back to file.nc, it will load the latest contents.  If I use the Workflow window and I switch files, it will re-load the files from disk each time you change, so this is just an issue with trying to re-load the file that is already loaded.

### How to reproduce

see "what happened"

### Operating System

MacOS 12.4 Monterey

breiler, 2022-07-17:
It *should* detect the file change, here is an example on an ancient Mac OS Sierra:
[image]

bertieconfundo, 2022-07-19:
ok, I see that *if the editor is open* then it will reload the file when it detects that it has changed.

I think that whether or not UGS reloads the file automatically when it's been written to should be the same regardless of whether or not the editor window is open.  I run UGS with the "Show editor when opening g-code files" option turned off, so this is why I wasn't getting this auto-reload behavior.

Also, I think that with no editor window, having UGS automatically reload the file from disk when it changes (after the initial open) is unexpected and potentially dangerous.

I think regardless of whether the editor window is open or not, the only time a file should be re-loaded is when a reload is explicitly asked for (and/or you open the same file).

Right now, if I open a file, (no editor open), change the file on disk, and then try to File->Open the same file again, UGS counterintuitively does nothing, rather than actually opening the file from disk that I told it to open.

What do you think?

closed 2023-08-26 by bertieconfundo