winder/Universal-G-Code-Sender #1930: FR: Add function to fully reset FluidNC controller

daxliniere, 2022-07-17:
### Description

In FluidTerm, CTRL+R will perform a full reset, which is occasionally needed (especially during all this beta testing of FNC and UGS).

The only ways we can currently do that are:
1) Disconnect in UGS, then load FluidTerm, send CTRL+R, exit, then reconnect with UGS.
2) Power cycle the controller.

It would be great if we could send that command from UGS. Perhaps it would be a part of the FluidNC protocol?

Thanks and all the best,
Dax.

breiler, 2022-07-17:
Once we establish a proper connection this button should be available. 

The problem is a design flaw in the FluidNC/GRBL protocol making it really hard to figure out which state it is in when connecting to it.  Right now there are two types of alarms, ex:
1. When connecting to a freshly powered controller it ends up in a alarm state the requires either homing or unlock.
2. When a limit pin has triggered a hard limit alarm which requires a soft reset, no other communication is possible.

Just enabling the soft reset button in this state does work but will be really user unfriendly. In a worst case scenario when you connect to a machine that has triggered a hard limit switch would look like this:
1. You connect to the controller
2. UGS shows that you are connected with the state "Alarm" and three buttons are enabled "Homing", "Unlock" and "Reset controller". These buttons should be enabled because it usually the normal operations to clear alarms.
3. You press "Unlock" but nothing happens (hard limit alarms doesn't allow it)
4. You press "Homing" but nothing happens (hard limit alarms doesn't allow it)
5. You press "Soft reset" and the controller reboots
6. UGS shows that you are connected but again with the state "Alarm", but this time this alarm is because the controller needs to be homed.
7. You press "Unlock" or "Homing" and the machine is now ready to use

In my opinion this is a really bad user interaction. I am leaning towards always resetting the controller when connecting which would always put the controller in a known state much like the old GRBL implementation.

daxliniere, 2022-07-17:
Yes, I agree. The multiple resets requirement makes it a but clunky and unintuitive to use.

Mooselake, 2022-08-07:
FWIW resets cause FNC to lose it's WCS position, it doesn't keep the offsets so rehoming won't recover.  I've been writing down work and machine coordinates at 0,0,0 but that's not a great solution.

breiler, 2022-08-08:
@Mooselake do you mean that this behaviour has since the legacy GRBL?

From the GRBL wiki:
```
0x18 (ctrl-x) : Soft-Reset

Immediately halts and safely resets Grbl without a power-cycle.
Accepts and executes this command at any time.
If reset while in motion, Grbl will throw an alarm to indicate position may be lost from the motion halt.
If reset while not in motion, position is retained and re-homing is not required.
An input pin is available to connect a button or switch.
```

If what you say is true there would not be any way to stop a running gcode program on FNC as that (AFIK) requires a soft reset to clear all buffers on the controller. I find this unlikely and I have not noticed this behaviour while developing support for FNC. So if you know something that I don't, please share.

daxliniere, 2022-08-09:
I've not seen this behaviour either.

Mooselake, 2022-08-09:
I thought it was strange also, but I was having a problem with what I thought was the current nightly from about a week ago.  It was frequently hanging requiring it to be killed and restarted.  Interestingly it continued sending gcode even though the UI had frozen.  After restarting a good percentage of the time (maybe always, long test session) the WCS offsets were lost, reset to zero.  In addition it caused the version of UGS that bobscnc recommended for their Revo (for their grbl controller) to die on startup, will chase that down later if necessary, guessing that's a config file issue.  I came across this issue while digging to find some info on FNC and UGS.

Turns out that what I'd actually downloaded was a very old version, right month and date but from 2020.   Does jfrog do that a lot?  Actually read the whole whack line and finally noticed the 2020, downloading again got 8/5/2022 (not 2020).  That actually has a FluidNC option, much better support for A (no macros required), and from limited testing works way better with FNC.  This was a cockpit error, let the frustration get to me, sorry to have wasted your time.   I'm relieved to be making some progress on this project.  This is the first time I've tried FluidNC, although I do have another XYZ router with grbl_esp32 on one of Bart's boards instead of this espduino clone.  First time XYA.  Bit of a journey.  Again, sorry, it was really another problem and soft reset appears to work as expected.   I'll test further today.

Mooselake, 2022-08-27:
My problem looks like it was related to an occasionally sticky home switch, which caused FNC problems when it remained closed after a pulloff and treated as a limit switch trip which caused a crash.  A recent pre-release suggests it's being worked on.  All seems to be well, including soft reset working as expected.

daxliniere, 2022-08-28:
Great news, @Mooselake. Unrelated to UGS, but have your tried the latest test build of FNC? We need as many people as possible to test the new event-driven codebase.

Mooselake, 2022-08-29:
I'm on 3.5.1, didn't try the two prerelease versions since the new switch fixed the sticky problem.  Maybe 10 hours of milling time on the Bobs Revolution with FNC and 3.5.0/3.5.1.  If you think it'll help I'll load the eventqueue2 version from a couple days ago (interesting that the releases are out of chronological order), and I could put a rubber band on the new switch to make it look stuck on, maybe pull off a wire to make it look stuck off too.

daxliniere, 2022-08-29:
Any test scenarios you can think of would be wonderfully helpful. (Sorry to hijack your forum, Joacim!)

Mooselake, 2022-08-30:
Sounds good.  I'll be OOT for a while

daxliniere, 2022-10-13:
Hey Joacim, any news on fixing the FNC 'can't connect on restart' bug? I often have to go into FluidTerm to CTRL+R just so UGS will connect to the controller again.

daxliniere, 2022-10-13:
This one specifically, see bottom of thread.
https://github.com/bdring/FluidNC/issues/613