winder/Universal-G-Code-Sender #1940: Rotary axis spiral rendering issue

reality-boy, 2022-07-22:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

Previously I reported issue 1864 (https://github.com/winder/Universal-G-Code-Sender/issues/1864) reporting that rotary axis rendering was coming out wrong.  This got fixed (thank you!), however now when trying to cut a spiral, so that both the x and a axis move in unison, the rendering shows the x axis move followed by the a axis move rather than a combined spiral move of both axis.

Here is the gcode to recreate this
G0 X0  Z5 A0 F1000
G1 X8 A80

And you can see snapshots of the issue at the link below.
https://hackaday.io/project/176110-multibot-cnc-v2/log/208841-ugs

### How to reproduce

See above, this is on the July 22-07-21th nightly build

### Operating System

Windows 10 x64 pro

breiler, 2022-07-22:
Thanks, I'll have a look!

breiler, 2022-07-22:
Should be available for you to try in the latest nightly build.

closed 2022-08-08 by breiler