winder/Universal-G-Code-Sender #1960: UGS Platform & classic version won't correctly load a gcode file with a blank first line

HoverClub, 2022-08-12:
### Version

2.0.11

### Hardware / Firmware

FluidNC

### What happened

If the 1st line in a gcode file is empty (line break only) then it isn't fully loaded.  The file displays in the editor OK but 'Loaded File:' shows 0 rows with no visualizer display and run does nothing.  If I remove the blank 1st line and save the file it all works as it should.  Same issue on both Classic and Platform UGS versions.
[image]
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9315993/messages.log)
 

### How to reproduce

Start UGS and open gcode file.

### Operating System

Linux Mint 20 & 21

### Anything else

Verified issue on two PCs (one with a clean Mint & UGS install).

breiler, 2022-08-12:
This should have been resolved in version 2.0.12

HoverClub, 2022-08-12:
> This should have been resolved in version 2.0.12

Yep - I confirm it is fixed.

closed 2022-08-12 by HoverClub