winder/Universal-G-Code-Sender #1962: fluidnc z probe module not working

dmbbender, 2022-08-14:
### Version

Other

### Hardware / Firmware

FluidNC

### What happened

I have a probe connected to my board that works with ugs 2.012 if I choose grbl but I am running fluidnc,  when I connect as fluidnc everything works great except the z probe?? 

### How to reproduce

connect a fluidnc board with a probe attached an try to use the probe module for z

breiler, 2022-08-15:
The probe function is not yet implemented. I could add the same code as with the GRBL controller but I have no way of testing it.

dmbbender, 2022-08-15:
I can test it if you would like. I have Fluidnc ver 3.4.4  if I connect as a grbl control and use the probe feature it works properly so I would expect it should work. I set up a macro for now!

breiler, 2022-08-15:
It should be available in the latest nightly build, please report back if it is working properly.

dmbbender, 2022-08-16:
It is not working correctly, sorry.  I am just going to use the macro G38.2 G91 F80 Z-20 P20; G00 Z10 M00  Thank you!

daxliniere, 2022-08-21:
@dmbbender, by the way, in latest FNC, you can use G53 in your z-probe macro. Do a search on the FNC Discord server for details.

nickmayer, 2023-04-16:
This was probably fixed by https://github.com/winder/Universal-G-Code-Sender/pull/2058. Let me know if that isn't the case.

breiler, 2024-02-05:
I have verified that it is working on my test rig with UGS 2.1.4 and FluidNC v3.7.11

closed 2024-02-05 by breiler