winder/Universal-G-Code-Sender #1979: window not showing macOS Ventura

karlhfr, 2022-09-10:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

when launching the app I can't see the window. it appears in the Mission Control preview but not on the desktop

none of the menu items open either
[image]
[image]

### How to reproduce

just open ups on a Mac with macOS Ventura beta

### Operating System

MacOS Ventura bete 4 & 5

### Anything else

it happens with all versions of bugs, .11, .10, .09

regevt, 2022-09-18:
can confirm, the same happend to me after upgrading to Ventura

breiler, 2022-09-20:
I don't own a Mac new enough to run MacOS Ventura, so if anyone want to help you are most welcome!

regevt, 2022-09-29:
I managed to build and run but it only runs the classic view not the "new" one
throw intellij I can run the "new" ui

what am I missing?

BTW the setup process for the env is rather complicated and require deprecated sdks, are you planing to move the project to a newer version of have and packages?

breiler, 2022-09-30:
Which SDK and packages do you mean is deprecated?

We still rely on `Java 13.0.1-9` which is old but as far as I know not deprecated. It is however currently the latest JDK that works on both MacOSX, Linux and Windows together with the latest version of the Jogamp-library. For ARM-processors we still need Java 8 because of some performance bugs introduced in Java 13.

I never got it to build with the latest Netbeans 15 framework so therefore we are currently stuck with Netbeans 12.2.

regevt, 2022-09-30:
I can run the software just fine if I start it from IntelliJ but if I run it manually the window is invisible

I am posting here the logs, maybe you can point me to a direction of a solution, otherwise I have no idea how to continue

`-------------------------------------------------------------------------------
>Log Session: Friday, September 30, 2022 at 8:19:47 AM Central European Summer Time
>System Info: 
  Product Version         = Universal Gcode Platform 12.2-d73d3dd898acc440c460924dabf8ad5feb90369f
  Operating System        = Mac OS X version 10.16 running on x86_64
  Java; VM; Vendor        = 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 13.0.1+9
  Java Home               = /Library/Java/JavaVirtualMachines/jdk-13.0.1+9-jre/Contents/Home
  System Locale; Encoding = en_NL (ugsplatform); UTF-8
  Home Directory          = /Users/regevtal
  Current Directory       = /Users/regevtal
  User Directory          = /Users/regevtal/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev
  Cache Directory         = /Users/regevtal
  Installation            = /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/ide
                            /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/bin
                            /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform
                            /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/etc
                            /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/java
                            /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/ugsplatform
                            /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/extra
                            /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform
  Boot & Ext. Classpath   = 
  Application Classpath   = /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/boot.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-modules.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-util-lookup.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-util-ui.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/lib/org-openide-util.jar
  Startup Classpath       = /Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/asm-tree-8.0.1.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/org-netbeans-libs-asm.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/org-openide-filesystems-compat8.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/org-openide-filesystems.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/asm-8.0.1.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/asm-commons-8.0.1.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/core.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/platform/core/core-base.jar:/Users/regevtal/Documents/Projects/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/ugsplatform/core/locale/core_ugsplatform.jar
-------------------------------------------------------------------------------
Warning: the fonts "Times" and "Times" are not available for the Java logical font "Serif", which may have unexpected appearance or behavior. Re-enable the "Times" font to remove this warning.
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading LocalizingService...
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Log location: /Users/regevtal/Library/Preferences/ugs/UniversalGcodeSender.json
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Loading settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying new command processor
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: RunFromService
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading JogService...
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: ContinuousJogWorker
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: AnalogJogAction
INFO [null]: Last record repeated 5 more times.
[... cut ...]  [org-netbeans-modules-navigator-ShowNavigatorAction.shadow, org-netbeans-core-io-ui-IOWindowAction.shadow, Separator3.instance, Classic, Plugins, com-willwinder-ugs-nbp-core-windows-DiagnosticsTopComponent.shadow, org-netbeans-core-windows-actions-ResetWindowsAction.shadow]
WARNING [org.openide.filesystems.Ordering]: Not all children in Menu/Help/ marked with the position attribute: [com-willwinder-ugp-welcome-WelcomePageTopComponent.shadow], but some are: [shortcuts.xml, Separator1.instance]
`

breiler, 2022-09-30:
No the logs didn't reveal anything. I think the way forward is to try and upgrade to the latest netbeans version. But that will take some time. I'll let you know if I manage to get it working.

breiler, 2022-09-30:
There is a fresh nightly build available here:
https://ugs.jfrog.io/ui/native/UGS/nightly/ugs-platform-app-ios.dmg

regevt, 2022-09-30:
still not working

larie438, 2022-10-31:
Any update on this?

totalementbarre, 2022-11-06:
Same issue for me,
solved with @breiler link !
thanks a lot !

regevt, 2022-11-06:
downloaded agin via the link, the app does work agin
thank you very mush

breiler, 2022-11-07:
Closing as users have reported that it is working

closed 2022-11-07 by breiler

bujinwang, 2022-11-23:
I have experienced the same problem with another java application. Can someone shine some lights on how to resolve this in code?

donedamned, 2022-11-24:
Same issue here after upgrading to Ventura. :c