winder/Universal-G-Code-Sender #1989: G-code sender 2.0.1.2 with CNC shield draw a wrong path

Shimidan, 2022-09-20:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

Sometimes, when I try to CNC a circle (or spine), it is ok, and sometimes not. For example, In the middle of a circle, the tip of the tool may go out of path and then return to path. Another example may occur in a multi press path:  some depths are ok and some not. In all cases the start and final point are exactly correct.

breiler, 2022-09-20:
Sample gcode and photos please.

This sounds like a problem with your hardware - such as stepper motors missings steps. 
But without more information it is impossible to answer.

Shimidan, 2022-09-20:
@breiler 
I attached it. The left image is not clear so I've colored the machine path in right image. See that the tip in a depth has gone to out of the path and then returned.
The start and final point are same, if there is a problem with step motors, these point were not same.
[image]

breiler, 2022-09-20:
Wow, real potato quality on the images. 😁 

Is the machine rigid enough, does the spindle flex if you apply pressure on it?

Shimidan, 2022-09-20:
Hard question, because i can move it 1 mm by pressure. I think there was a way to test " Is it enough? " for example test of the problem Vs speed of X-Y movment.

Shimidan, 2022-09-21:
> Wow, real potato quality on the images. grin
> 
> Is the machine rigid enough, does the spindle flex if you apply pressure on it?

I used some adhesive and now it ok. I hope it will be ok, too.

closed 2022-10-01 by breiler