winder/Universal-G-Code-Sender #1997: When I Jog x y the status seem to remain green Jogging

cpapplefamily, 2022-09-30:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

When I Jog x y the status seem to remain green Jogging
[image]

### Operating System

Win 10

### Anything else

Every time I jog x or y the system jogs and does follow the next jog request but the UI seems to lock up

breiler, 2022-10-01:
Can we have your log file from here: `C:/Users/[your username>]/.ugsplatform/2.X/dev/var/log/messages.log`

cpapplefamily, 2022-10-01:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9690349/messages.log)

breiler, 2022-10-01:
The log says it is out of memory. Please try the 64-bit Windows variant, version 2.0.12.

breiler, 2022-11-01:
Closing as we did not get any response up from the user. If anyone else has this problem, make sure you have downloaded the 64-bit edition of UGS from here: https://github.com/winder/Universal-G-Code-Sender#downloads

closed 2022-11-01 by breiler