winder/Universal-G-Code-Sender #1999: No USB/Serial port option on OSX

ArketypeDesign, 2022-10-03:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

I have been running UGS successfully  on a Raspberry Pi3, but performance left a little to be desired, and I had access to a "retired" 2011 MacMini which is much snappier.

[image]

I cannot select a USB/ Serial port from UGS on the Mac, I only see "bluetooth" options.
I have opened the Sender options and set JSerialComm.

I am not running any other versions of Java.

i have tried both 2.012 and 2.09 versions of UGS with similar results.
Any help would be appreciated.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/9693836/messages.log)

### Operating System

OSX 10.13.6 High Sierra

breiler, 2022-10-03:
Depending on what controller you are using you need drivers for it to be identified by the OS. But you need to know which USB chip it is using. In my case I have a couple of cheap chinese boards that uses CH340.

breiler, 2022-11-01:
Closing as we didn't get any response from the user. This is also related to the hardware and its driver rather than UGS so there is not much we can do about it.

closed 2022-11-01 by breiler

ArketypeDesign, 2022-11-01:
Installed driver, and it is working! Thanks!