winder/Universal-G-Code-Sender #2007: Unable to download

niteshkadyan, 2022-10-15:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

Unable to download. https://ugs.jfrog.io/ugs/UGS/v2.0.12/ugs-platform-app-linux.tar.gz redirects to https://ugs.jfrog.io/ui/login/ fof login

### How to reproduce

Click on the download link

### Operating System

Linux

breiler, 2022-10-15:
There is something wrong with jfrog again, please use the download links on the bottom part of this page:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.12

Witawat, 2022-10-19:
> มีบางอย่างผิดปกติกับ jfrog อีกครั้งโปรดใช้ลิงค์ดาวน์โหลดที่ส่วนล่างของหน้านี้: https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.12

can't download file

breiler, 2022-10-19:
@Witawat on the bottom part of that page, under the section assets.

peterfalm, 2022-11-26:
Thank you so much, I have been really struggling to install UGS and now at last I have managed it thanks to you.
I don't know what's going on with that Frog site but it's a waste of time and so much time wasted trying to get that to work. 😀

Tjhcncuser, 2022-11-29:
I'm also having trouble.
Which file at the end do I download?
After extracting which file do I select to install?
Thx

breiler, 2022-11-30:
I have now updated all download links to use github instead of JFrog:
https://github.com/winder/Universal-G-Code-Sender#downloads

closed 2022-11-30 by breiler