winder/Universal-G-Code-Sender #2008: How to download?

GrahamJB, 2022-10-15:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

I am trying to download UGS, the latest stable version, but when I do so I am taken to a website called JFrog which requires me to register, is there any way just to download the version from Github without giving my details to a third party who I have no reason to trust?

### How to reproduce

Try and download from Github

khturnings, 2022-10-15:
Check previous issues - see https://github.com/winder/Universal-G-Code-Sender/issues/2007

GrahamJB, 2022-10-16:
Thank you, I thought I had but didn't scroll right down to the bottom.

closed 2022-10-16 by GrahamJB