winder/Universal-G-Code-Sender #2009: Window Controls not working on MacOS Ventura beta

sricks1, 2022-10-17:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

Launching UGS platform on Mac OS Ventura Beta, the main window draws, and you can click on a menu, it highlights, but nothing else.  Same with dropdown menus in the main window. Hitting the Port Sync button displays 1 item in the list, but drop-down does nothing.  Same with Baud.

### How to reproduce

Just open in MacOS Ventura Beta.

### Operating System

MacOS 13.0 Beta (22A5373b)

### Anything else

Occurs every time.  After a while nothing happens and window freezes.  Under Force Quit the app still show as running.  But the only way to quit is to Force Quit.

fehguy, 2022-12-23:
Try the nightly. `20221222` works for me on 13.0.1

breiler, 2023-01-27:
According to #2036 it should be working better with the 2.0.13 version

closed 2023-01-27 by breiler