winder/Universal-G-Code-Sender #2015: Unable open on Raspberry Pi 4

motaviegas, 2022-10-21:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

I have tried to run the file without any success, I have tried many RPi OS versions, even Ubuntu 64 / 32 and in none UGCS opens. I have given execution permissions but nothing happens.

### Operating System

RPi OS

sepiroth887, 2022-11-15:
don't pull the version with self-executing binaries. pull the universal app and extract, install jdk >=8 < 13 (i.e. openjdk-11-jdk), make sure java -version shows the correct version and run the binary in ugs/.bin/ugsplatform  (bear in mind, the latest nightly is what you want to pull to get the new opengl implementation pulled in as well)

breiler, 2023-01-23:
I could not get it to work reliably using `Raspberry Pi OS 64-bit`, mouse clicks and menus behaved funky and the application ran with a high CPU-load. Using `Raspberry Pi OS 32-bit` I run our prepackaged ARM-variant of UGS nightly without any modifications in the OS.

breiler, 2023-10-15:
There is now a 64-bit ARM version in the nightly builds that you can try out.