winder/Universal-G-Code-Sender #2016: XY scale error when moving the spindle

Meina88, 2022-10-21:
### Version

2.0.12

### Hardware / Firmware

GRBL 1.1

### What happened

Hello! I am using UGS on my CNC router. The hardware is a development of my property based on X-Controller. I have had no problems so far. What happens is that moving the spindle in "Y" and "Z" direction everything works fine. But moving the spindle in the X direction only moves half the distance. This only happens when I use the Joistick. When executing a G code everything works fine.
[image]

### How to reproduce

I really don't know how to recreate the situation. But I think there may be some configuration problem that I don't identify.

### Operating System

Windows 11 Home Windows Feature Experience Pack 1000.22000.978.0

### Anything else

This problem only occurs occasionally. But now it has become permanent on my router. It only happened to me once before. But now it appeared again.

breiler, 2022-10-25:
What feed speed are you using? The maximum feed rate is higher for the X axis ($110), could the stepper motor be stalling?
Does the same happen if you jog at a speed of 500mm/min?

Meina88, 2022-10-25:
Hello! thanks for your answer. I finally figured out part of the problem. I clarify that this problem appeared on a machine that I sold to a client.

-First of all, I went to his workshop, tried several versions of UGS and two laptops. The problem persists. That means the problem was with the machine.

- I took the control cabinet to my workshop and tried it with another machine that uses the same motors and the problem disappeared.

- That makes me think that the problem is in the stepper (or in the cable that goes to the stepper or connector). But if the problem was a motor I have a doubt. Why does it only do this when sending commands through Jog and not when sending G code? This is what has me a bit lost. I'll keep working on it while I wait for your comments.

I am very grateful, Pablo Meinardo

breiler, 2022-10-26:
As mentioned, my suspicion is that your X motor is stalling, that is why I asked what feed rate you are jogging in.

If your gcode program is running at a max 1000mm/min but you are jogging at a max 2000mm/min, it could mean that the max rate for the X-axis is somewhere in between.

Does the problem occur when trying to jog at 500mm/min?

breiler, 2023-01-19:
Closing as we got no response

closed 2023-01-19 by breiler