winder/Universal-G-Code-Sender #2023: USG Won't completely open

DTinfl, 2022-10-29:
### Version

2.0.9

### Hardware / Firmware

GRBL 0.8

### What happened

I have been using UGS for a few months with Windows 10 and an X-Controller.  I don't think I changed anything and now when I open UGS it starts to open and gets to the part where Loading Module Services and then the UGS opening page just disappears and it stops opening.  I have reloaded different versions of UGS and restarted Windows several times and the exact same thing happens. 

### How to reproduce

Click on the USG platform file, click on bin, click on the USG platform 64 version.  The opening screen comes up, you can see the different parts loading and when it gets to the Loading Module Services part everything disappears.   

### Operating System

Windows 10, I have tried two different update versions

breiler, 2022-10-30:
The configuration might have become corrupt.

Try deleting the files under `C:/users/[your username>]/.ugsplatform/2.X/dev/var/cache`
If that doesn't work try deleting the `C:/users/[your username>]/.ugsplatform`

DTinfl, 2022-10-31:
Thanks for the reply.  I could not find these files so I delete everything I could find with ugsplatform in it except the original folder that did open in the past.  After doing that the original file opened.  I don't know what happened or how it happened but it would appear there was a corrupt file somewhere that was keeping it from working.  Again, thanks.  It appears I am now up and running again.

breiler, 2022-10-31:
This can happen when upgrading UGS. Not certain on why it happens though. I am glad that it is working agin.

closed 2022-10-31 by breiler